zacharee/Bifrost #1: Notification issue

random-tester-nuked, 2021-03-01:
Hi thanks for your amazing work,
I've found a little issue (I'm on Samsung S10+ wirh last firmware android 11 one ui 3.1)

So when the firmware aren't available to download one notification stuck at same point eternally

[image]
[image]

zacharee, 2021-03-01:
The notification will stay as long as the app is running.

random-tester-nuked, 2021-03-01:
Oh ok thnx but even if I close the apps from recent apps the notifications are still here

random-tester-nuked, 2021-03-01:
Thanks you for all

zacharee, 2021-03-02:
https://github.com/zacharee/SamloaderKotlin/releases/tag/0.2.1

closed 2021-03-02 by zacharee