zacharee/Bifrost #2: New Name

zacharee, 2021-03-07:
Samsung Firmware Downloader is a bad name. It's too generic and too long.

If anyone has a suggestion, leave a comment.

mohammad92, 2021-03-08:
Also spaces make issue with linux.
samfirmloader, samfirmloaderkot. sflk.

david-lev, 2021-03-13:
Amazing tool by the way... 
Suggestions:
- Samfirm
- Samloader
- downsamfirm

svrselja, 2021-03-13:
FirmUp
UpSam 
UpFirmFinder
UpGalaxy
Galaxy Flash data
Samsung Files
SamUpData

Frivolous2, 2021-03-13:
Bifrost, after the Norse mythology bridge that connects the realm of mortals to the realm of the gods.

JET-6, 2021-03-14:
Sam's Firm D

roynatech2544, 2021-03-14:
Samloaderkt

zacharee, 2021-03-14:
> Bifrost, after the Norse mythology bridge that connects the realm of mortals to the realm of the gods.

This seems like the best option so far. 

I'll wait a bit longer to see if there's another suggestion I like, though.

pfhayter, 2021-03-14:
Quasar, after flashing lights in our galaxy and beyond.

0n1cOn3, 2021-03-14:
Saphirm - Pronounciation from Sa(m) and Saphir + Firm(ware)

Saphir as Contribution to get Samload on the GUI Side. + Cross Platform Support.

Frivolous2, 2021-03-15:
SamFirm, like XperiFirm https://xperifirm.com/

JohnSmithGoogl01, 2021-03-17:
Samdicate

tcharding, 2021-04-06:
-1 for Bifrost (unless its tonuge in cheek). Implies we are the mortals and Samsung the gods - after trying for many days to flash a Samsung device I'd prefer a name that implies Samsung is from the underworld. The fact that we even need to write a tool to connect to them emphasizes my point.

tcharding, 2021-04-06:
Suggestion: `samfer` (Samsung Firmware Downloader)

LuuchoRocha, 2021-07-11:
ZSloader

CiprianDinca9, 2021-07-13:
Suggestion: SamFirmStore.

MLXProjects, 2021-08-23:
What about SaphirDL? Easily pronounceable and reflects the original name (the PH instead of a F is because of the pronunciation)

Githubraum, 2021-11-05:
suggestion: Samdowner
short: Samdo

CiprianDinca9, 2021-11-05:
SaphirDL. I agree with this. It feels like the best option so far.

viktorstalzkov, 2021-11-21:
SamsFirmLoad

GoodSalad

SolidSandwich / SpringSammich / SamsSub

zacharee, 2021-12-07:
With version 1.0.0, Samsung Firmware Downloader has been renamed to Bifrost.

closed 2021-12-07 by zacharee