zacharee/Bifrost #3: macOS Maintainer

zacharee, 2021-03-07:
Since I don't have a Mac, I can't make macOS builds.

If anyone with a Mac is willing to maintain macOS builds, please let me know. It should be a temporary arrangement, since JetBrains is planning to add cross-compilation support in the future.

ktetreault, 2021-03-13:
I have a working Hackintosh environment I could help out with?

zacharee, 2021-03-13:
Would you mind building to see if everything works?

ktetreault, 2021-03-13:
I should be able to follow some instructions later on today or tomorrow.

Kevin Tetreault
(727) 712-6716
***@***.***

0n1cOn3, 2021-03-14:
I have a older Macbook but atm running without any workload (Standby most of the time). XCode and some other dev utils are already installed.

zacharee, 2021-03-14:
You need Android Studio to make the DMG.

DQuealey, 2021-03-14:
Go Zach, 
Hey, i'm not a developer but very interested in your firmware download tool.. I did'nt find a way to contact you privately so i'll post here for now.. 
I downloaded 3.0.1 on my windows 7 pro pc.. Got an error message this program isn't compatible with my system.. Contact publisher to find out if it works with a 32 or 64 bit enviorment. 
Please advise as i'd love to have your program.. I'm very nervous downloading firmware from strange links. 
Trying to find firmware for my SM-G935A. Already bit the dust one one $600.00 S7.
Thanks for the great work! 
Darrell.

zacharee, 2021-03-14:
@DQuealey can you start a new issue?

ed789d0, 2021-03-14:
Could you maybe use GitHub actions for building it? It supports macOS workflows.

DQuealey, 2021-03-14:
> @DQuealey can you start a new issue?

Just started an issue thanks Zach.

zacharee, 2021-03-14:
@ed789d0 I set up an action that seems to successfully build a DMG. 

It isn't signed or notarized, but I think it's still possible to run it with these steps:
- Download it from the [workflow page](https://github.com/zacharee/SamloaderKotlin/actions/runs/651819294) (under Artifacts at the bottom).
- Extract the ZIP and then open the DMG.
- If there's a security error, hit "Cancel" and then go to System Preferences --> Security. Hit the lock icon and then hit the "Allow Anyway" button. (There's a process with images [here](https://github.com/hashicorp/terraform/issues/23033#issuecomment-542302933)).
- Then it should be possible to open the app.

0n1cOn3, 2021-03-14:
Alright - Iv tested the macOS Version. It seems to be, that it won't work... It shows as "Damaged Package".
Sometimes you can bypass that when you run the executable via terminal... But there is another problem... After change the permission to run it this happen:

Error opening "/Applications/Samsung Firmware Downloader/Contents/Contents/app/Samsung Firmware Downloader.cfg"

Theres twice "Contents" in the binary.

zacharee, 2021-03-15:
@0n1cOn3 I'm looking at the DMG for 0.3.2, and it doesn't have a nested Contents directory.

zacharee, 2021-03-15:
I also just tested the app in a Big Sur VM and it (surprisingly) opens with any security popups. I don't think launching directly from a binary is supported for macOS apps.

zacharee, 2021-03-15:
@0n1cOn3 can you try this:

- Copy the .app to Applications.
- Open Terminal and run `sudo xattr -rds com.apple.quarantine /Applications/Samsung\ Firmware\ Downloader.app/`.
- Try opening the app now.

I tested this successfully on a real Catalina Mac.

DQuealey, 2021-03-16:
Zach, you do understand I'm running in a windows 7 environment right? I just couldn't find out where to send you a message besides the mac section..

I don't know how to run commands and whatnot I'm not computer savvy I need exe programs to setup and run....lol

Sent via the Samsung Galaxy S7 edge, an AT&T 4G LTE smartphone

zacharee, 2021-03-16:
> Zach, you do understand I'm running in a windows 7 environment right? I just couldn't find out where to send you a message besides the mac section.. I don't know how to run commands and whatnot I'm [...]

I'm not replying to you. When you posted in this issue, you were subscribed, and so you get notifications for replies.

0n1cOn3, 2021-03-18:
> @0n1cOn3 can you try this:
> 
> 
> 
> - Copy the .app to Applications.
> 
> - Open Terminal and run `sudo xattr -rds com.apple.quarantine /Applications/Samsung\ Firmware\ Downloader.app/`.
> 
> - Try opening the app now.
> 
> 
> 
> I tested this successfully on a real Catalina Mac.

This Step is a workaround. This might be gone in the future. If this is the "Solution". It could be add to the Readme.MD 😀

Gonna try that asap.

0n1cOn3, 2021-03-18:
> I also just tested the app in a Big Sur VM and it (surprisingly) opens with any security popups. I don't think launching directly from a binary is supported for macOS apps.

It is supported. It won't run at all if it is not that case😁

closed 2021-04-27 by zacharee