zacharee/Bifrost #4: Application runtime error

gidano, 2021-03-14:
Hi,
Unfortunately, I can't run the app. I attach a screenshot and log file, I want to ask for help. I can't resolve the error...

[image]
[hs_err_pid2988.log](https://github.com/zacharee/SamloaderKotlin/files/6136344/hs_err_pid2988.log)

zacharee, 2021-03-14:
I've reported this issue to the JetBrains team to see if they know what's happening. They might need additional information, Would you mind subscribing to the issue and responding if needed?

https://github.com/JetBrains/skiko/issues/72

qbrian, 2021-03-14:
@gidano 
I get the same and even worse because on attempt 2 nothing came out and the solution was to run it as administrator
[image]

gidano, 2021-03-14:
Thanks, I tried but the phenomenon is the same!

gidano, 2021-03-14:
> I've reported this issue to the JetBrains team to see if they know what's happening. They might need additional information, Would you mind subscribing to the issue and responding if needed?
> 
> [JetBrains/skiko#72](https://github.com/JetBrains/skiko/issues/72)

Of course, I'll help if I can.

qbrian, 2021-03-14:
@gidano,  I understand, it's fine, perfect

-regards [ @gidano , @zacharee ]

gidano, 2021-03-15:
I was hoping this morning that v0.3.2 would work for me too. But the same phenomenon as in previous versions.

qbrian, 2021-03-15:
Oh :c 
Hello, I understand, did you try to update the graphics and java drivers, and those determined by windows and windows itself?

qbrian, 2021-03-15:
look here https://github.com/zacharee/SamloaderKotlin/compare/0.3.2...master#commits_bucket

gidano, 2021-03-16:
I have never had a problem with my Intel video card, the latest driver is installed for every device.
Now turning off Intel has solved the problem, starting the application using AMD Radeon.
Regardless, I think it should work smoothly with the internal Intel card as well, its use has never been a problem. I just work on the machine, I don't play... :)
Thanks to everyone who tried to help!
[image]

closed 2021-03-20 by gidano