zacharee/Bifrost #5: Windows 7 problem.

DQuealey, 2021-03-14:
When installing V 3.0.1, Windows 7 pro. Says this program isn't compatible with this version of windows.. It apparently can't tell if it's a 32 or 64 bit program.

zacharee, 2021-03-14:
Are you on a 32-bit or 64-bit version of Windows?

zacharee, 2021-03-14:
I'm going to guess it's 32-bit, since I get a similar error with a 32-bit virtual machine. Unfortunately, JetBrains Compose doesn't currently support 32-bit builds.

closed 2021-03-14 by zacharee

DQuealey, 2021-03-15:
I do believe it is a 32 bit system Zach.. Bummer! Maybe in future updates you'll make it 32 compatable?

I'm sure a lot of us would enjoy this work!

Darrwll.

Sent via the Samsung Galaxy S7 edge, an AT&T 4G LTE smartphone

DQuealey, 2021-03-15:
Yes, Sir....

Sent via the Samsung Galaxy S7 edge, an AT&T 4G LTE smartphone

zacharee, 2021-03-15:
There isn't anything I can do directly.

DQuealey, 2021-03-16:
I understand

Sent via the Samsung Galaxy S7 edge, an AT&T 4G LTE smartphone