zacharee/Bifrost #10: Great for downloading current firmware but be careful downloading from history on Windows

dazemc, 2021-04-07:
Windows 10 x64 
I was attempting to download an older firmware version and it appeared the correct version was downloading. I flashed the files and the version didn't change. I had a mini heart attack because I thought I had inadvertently updated a revision on my bootloader, thus loosing my unlock token. Well come to find out, this tool downloaded the most recent firmware but then during decrypting it changed the filename to the version I downloaded. So the decrypted file is named *CUA2*.zip but when I unzip it, its the *DUBA*.tar

Attached is log
[hs_err_pid9828.log](https://github.com/zacharee/SamloaderKotlin/files/6274815/hs_err_pid9828.log)

dazemc, 2021-04-07:
I just realized that log is probably useless. Is there a logfile for samloader or a way I can enable it?

zacharee, 2021-04-07:
There's no logging built in right now. I'll look into why the firmware is wrong, but I'm going to guess Samsung is serving it incorrectly. Is this the Note20 Ultra?

dazemc, 2021-04-07:
No, it's a Galaxy S20+ (SM-G986U1)

zacharee, 2021-04-27:
Which region are you using?

dazemc, 2021-04-27:
XAA

zacharee, 2021-04-27:
From what I can tell, Samsung is just serving the latest firmware no matter what is specified in the request. I think this is new.

TheAirBlow, 2021-09-03:
I use original samloader and when specifying a firmware it downloads the requested one.
You use some other way to get the files, because I am downloading two firmware versions and their file sizes are totally different.

zacharee, 2021-09-04:
> I use original samloader and when specifying a firmware it downloads the requested one.
> You use some other way to get the files, because I am downloading two firmware versions and their file sizes are totally different.

The method I use is directly ported from Samloader. It was a server side change from Samsung to only serve the latest firmware, at least for most devices. Can you tell me which model and CSC you used? Samsung may have reverted the change.

TheAirBlow, 2021-09-04:
SM-A207F/SER

SlackingVeteran, 2021-09-05:
> The method I use is directly ported from Samloader. It was a server side change from Samsung to only serve the latest firmware, at least for most devices. Can you tell me which model and CSC you [...]

Hi, Frija dev here. Its always been this way. Samsung never serves older firmware unless you happen randomly to find a specific xml command which only Samsung devs have access to. I am pretty sure you know which xml command I am talking about (Get->CmdId>2>LATEST_FW_VERSION), so far no one knows what other possible commands Samsung uses internally because every version of Kies or SmartSwitch (Windows/MacOS) uses that command from above. Maybe if someone gets hand on Fenrir (Internal tool from Samsung which does everything from Device management, Firmware download, Firmware install, FRP bypass, you name it) we might hit a jackpot

SlackingVeteran, 2021-09-05:
Just realized you dont even use `Get` command

TheAirBlow, 2021-09-05:
SM-A207F/SER downloaded using Samloader
```
 theairblow@theairblow > cd Samsung/btk1-10/bl 
 theairblow@theairblow > md5sum aboot.mbn     
04f83d857c5575d6b9dc772c97fb6deb  aboot.mbn
 theairblow@theairblow > cd ../../cufa-11/bl
 theairblow@theairblow > md5sum aboot.mbn   
be107d5bd8cd377ccb66d0a0f7c4582d  aboot.mbn
```

Hashes are different

TheAirBlow, 2021-09-05:
> > > I use original samloader and when specifying a firmware it downloads the requested one.
> > > You use some other way to get the files, because I am downloading two firmware versions and their file sizes are totally different.
> > 
> > 
> > The method I use is directly ported from Samloader. It was a server side change from Samsung to only serve the latest firmware, at least for most devices. Can you tell me which model and CSC you used? Samsung may have reverted the change.
> 
> Hi, Frija dev here. Its always been this way. Samsung never serves older firmware unless you happen randomly to find a specific xml command which only Samsung devs have access to. I am pretty sure you know which xml command I am talking about (Get->CmdId>2>LATEST_FW_VERSION), so far no one knows what other possible commands Samsung uses internally because every version of Kies or SmartSwitch (Windows/MacOS) uses that command from above. Maybe if someone gets hand on Fenrir (Internal tool from Samsung which does everything from Device management, Firmware download, Firmware install, FRP bypass, you name it) we might hit a jackpot. Fenrir app is locked behind auth and is tied to MAC address of PC and is only provided to Samsung authorized repair centers, so I doubt anyone will ever get to use it without someone authorized by Samsung wanting to help reverse engineer the App. Fenrir was made after Odin was leaked in the wild which required no authentication.

Even getting it's files will be something: we can try to bypass any of the restrictions, reverse-engineer it's binaries. It will just be fun to play with.

SlackingVeteran, 2021-09-05:
> Even getting it's files will be something: we can try to bypass any of the restrictions, reverse-engineer it's binaries. It will just be fun to play with.

Thing is you can't reverse engineer Fenrir, it uses Themida which disables reverse engineering its binaries and libraries, soon as you attach debugger or use reflection it crashes everything. They thought about the reverse engineer possibility which is why I think macOS version doesn't exist.

SlackingVeteran, 2021-09-05:
Using wireshark while Fenrir is requesting firmware and downloading is only thing you can try so that you could see what kind of requests they make to get older firmware but again you can't use Fenrir outside assigned PC for Samsung authorized repair shops. I got to use it once back in 2018 for like 1 minute when Samsung sent someone to repair my TV at home. Repair man let me play around for lil bit but that was it couldn't do anything I really wanted to do.

TheAirBlow, 2021-09-05:
> Thing is you can't reverse engineer Fenrir, it uses Themida (https://www.oreans.com/Themida.php) which disables reverse engineering its binaries and libraries, soon as you attach debugger or use [...]

MAC address check not sounds so promising and MAY be hacked. Crazy shit, I know.

SlackingVeteran, 2021-09-05:
> > > > > > I use original samloader and when specifying a firmware it downloads the requested one.
> > > > > > You use some other way to get the files, because I am downloading two firmware versions and their file sizes are totally different.
> > > > > 
> > > > > 
> > > > > The method I use is directly ported from Samloader. It was a server side change from Samsung to only serve the latest firmware, at least for most devices. Can you tell me which model and CSC you used? Samsung may have reverted the change.
> > > > 
> > > > 
> > > > Hi, Frija dev here. Its always been this way. Samsung never serves older firmware unless you happen randomly to find a specific xml command which only Samsung devs have access to. I am pretty sure you know which xml command I am talking about (Get->CmdId>2>LATEST_FW_VERSION), so far no one knows what other possible commands Samsung uses internally because every version of Kies or SmartSwitch (Windows/MacOS) uses that command from above. Maybe if someone gets hand on Fenrir (Internal tool from Samsung which does everything from Device management, Firmware download, Firmware install, FRP bypass, you name it) we might hit a jackpot. Fenrir app is locked behind auth and is tied to MAC address of PC and is only provided to Samsung authorized repair centers, so I doubt anyone will ever get to use it without someone authorized by Samsung wanting to help reverse engineer the App. Fenrir was made after Odin was leaked in the wild which required no authentication.
> > > 
> > > 
> > > Even getting it's files will be something: we can try to bypass any of the restrictions, reverse-engineer it's binaries. It will just be fun to play with.
> > 
> > 
> > Thing is you can't reverse engineer Fenrir, it uses Themida (https://www.oreans.com/Themida.php) which disables reverse engineering its binaries and libraries, soon as you attach debugger or use reflection it crashes everything. They thought about the reverse engineer possibility which is why I think macOS version doesn't exist.
> 
> MAC address check not sounds so promising and MAY be hacked. Crazy shit, I know.
> Also, it should be not MAC address because it is networking stuff and can be easily changed, and it is per-network controller

[image]

TheAirBlow, 2021-09-05:
> [image]

We need to save all info we have about this software somewhere. And about samsung device protocols, it's servers protocols in general

TheAirBlow, 2021-09-05:
It is just Scamsung, what would you expect? For example, my phone (SM-A207F / Galaxy A20s) doesn't accept any custom binary, [this](https://forum.xda-developers.com/t/a207f-a20s-rooting-twrp.4329011/) and [this]((https://forum.xda-developers.com/t/guide-twrp-root-how-to-install-twrp-and-root-on-a20s-sm-a207f.4293965) for more information.

zacharee, 2021-09-07:
> SM-A207F/SER

It looks like this device is an exception. Samsung's servers are providing older firmware for it. But using SM-N986U/TMB, for example, will serve the latest firmware no matter what.

TheAirBlow, 2021-09-07:
They provide old firmware for this phone, but completely lock the
bootloader on Exynos model? Thanks, samsung.

TheAirBlow, 2021-11-28:
> > SM-A207F/SER
> 
> It looks like this device is an exception. Samsung's servers are providing older firmware for it. But using SM-N986U/TMB, for example, will serve the latest firmware no matter what.

Actually, it server only the last **two** firmware versions available.
Output of [Syndical](https://github.com/TheAirBlow/Syndical):
```
Device: SM-A207F/SER
Connecting to FUS server...
┌─────────────────────────────────────────────────────────┬────────────────┬────────────┬────────┐
│ Version                                                 │ Android        │ Size       │ Latest │
├─────────────────────────────────────────────────────────┼────────────────┼────────────┼────────┤
│ A207FXXU2CUI2/A207FOXM2CUI2/A207FXXU2CUI2/A207FXXU2CUI2 │ R(Android 11)  │ 4556071616 │ True   │
│ A207FXXU2BTK1/A207FOXM2BTK1/A207FXXU2BTK1/A207FXXU2BTK1 │ Q(Android 10)  │ 3677711232 │ False  │
│ A207FXXU2BTD7/A207FOXM2BTD8/A207FXXU2BTD7/A207FXXU2BTD7 │ Q(Android 10)  │ 3677711232 │ False  │
│ A207FXXU1ASJ5/A207FOXM1ASJ5/A207FXXU1ASJ5/A207FXXU1ASJ5 │ Pie(Android 9) │ 3677711232 │ False  │
│ A207FXXU2BTH2/A207FOXM2BTH1/A207FXXU2BTH1/A207FXXU2BTH2 │ Q(Android 10)  │ 3677711232 │ False  │
│ A207FXXS2ASL3/A207FOXM2ASL3/A207FXXS2ASL3/A207FXXS2ASL3 │ Pie(Android 9) │ 3677711232 │ False  │
│ A207FXXU2BUD4/A207FOXM2BUD5/A207FXXU2BUC1/A207FXXU2BUD4 │ Q(Android 10)  │ 4556071616 │ False  │
│ A207FXXU2CUH5/A207FOXM2CUH5/A207FXXU2CUH5/A207FXXU2CUH5 │ R(Android 11)  │ 4556071616 │ False  │
│ A207FXXU2BTE1/A207FOXM2BTE2/A207FXXU2BTE1/A207FXXU2BTE1 │ Q(Android 10)  │ 3677711232 │ False  │
│ A207FXXU1ASI2/A207FOXM1ASHI/A207FXXU1ASHI/A207FXXU1ASI2 │ Pie(Android 9) │ 3677711232 │ False  │
│ A207FXXU2BUD2/A207FOXM2BUD2/A207FXXU2BUC1/A207FXXU2BUD2 │ Q(Android 10)  │ 3677711232 │ False  │
│ A207FXXU2ATB1/A207FOXM2ATB1/A207FXXU2ATB1/A207FXXU2ATB1 │ Pie(Android 9) │ 3677711232 │ False  │
│ A207FXXU2CUFA/A207FOXM2CUFB/A207FXXU2CUFA/A207FXXU2CUFA │ R(Android 11)  │ 4556071616 │ False  │
│ A207FXXU2BTI1/A207FOXM2BTI1/A207FXXU2BTH4/A207FXXU2BTI1 │ Q(Android 10)  │ 3677711232 │ False  │
└─────────────────────────────────────────────────────────┴────────────────┴────────────┴────────┘
                                                                                      
Fetching firmware information ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 100% 00:00:10
```

TheAirBlow, 2021-11-28:
@SlackingVeteran, I have a question for you.
Why do you still use Samsung's DLLs which makes Frija non-crossplatform?
It was already reverse-engineered and works fine.
Also, can you provide all information you currently know about Samsung FUS endpoints and Fenrir?

TheAirBlow, 2022-05-31:
@SlackingVeteran 
https://www.tamiraat.com/repository/other/1398/10/09/efmbx0ux.bs0.pdf
Why the fuck a how-to PDF is public? It has fenrir/odin guides.

TheAirBlow, 2022-10-30:
> > No, it's a Galaxy S20+ (SM-G986U1)
> > […](#)

Samsung probably did it intentionally to save on space and just to give a middle finger to people who want older firmware.

BillAnt1, 2023-08-07:
> Using wireshark while Fenrir is requesting firmware and downloading is only thing you can try so that you could see what kind of requests they make to get older firmware but again you can't use [...]
.
.
I just read today that Samsung has designated most YouBreakIFix stores as authorized repair shops.  So, that may open the possibility of capturing the Fenrir packets between the phones, computer, and server if we can find a "friendly tech" in one of those shops. ;)
.
https://www.ubreakifix.com/samsung-repair#:~:text=uBreakiFix%20is%20a%20Samsung%20Authorized,we%20can%20fix%20it%20all.

AbuJaafar88, 2023-11-16:
###

maled6, 2025-01-01:
"[MODEL] IMEI"

L3P3, 2025-01-09:
Just loaded the old version file from samfw.com, it worked, not the newest version! :grin:

SXW617746893, 2026-10-16:
It always stops downloading after reaching 90%