zacharee/Bifrost #11: Pause button

abiud254, 2021-04-11:
The resume feature is _alright_ but i don't trust files that stopped downloading because of an error; something always tells me they're _corrupted_ so I don't think a pause button would hurt anyone. But **overall** it's very **awesome** and would recommend it to any other firmware source out there. **Keep up the good work.**

zacharee, 2021-04-12:
The files are downloaded sequentially one byte at a time. Unless something goes really wrong on the operating system end, it shouldn't be possible to have corruption. The pause button would effectively be the same as closing the program.

abiud254, 2021-04-14:
Thanks for the reply. I actually tried turning off the pc then resuming the download by reopening the program..still worked and I actually used the file and it wasn't corrupted. Nice program.

closed 2021-04-27 by zacharee