zacharee/Bifrost #17: The os does not support: 'Taskbar.setIconImage' message (but tool works well - Windows 10 Pro, 64bit)

7wells, 2021-06-27:
Hello!

I find your tool very useful and it works. Thank you! :-) There is just a minor thing I would like to make you aware of. In Windows 10 (Pro, 64bit), a window opens when executing your tool, showing the below message:
> The os does not support: 'Taskbar.setIconImage'

However, the main window of your tool open separately, and everything seems to work just fine, irrespective of this message. I don't know where this comes from (and it seems to be of minor importance), but I thought you might be interested in this.

If you need more info, please let me know.

zacharee, 2021-06-27:
Don't worry about that message. It's just for debugging purposes.

closed 2021-06-27 by zacharee

GravuTrad, 2021-09-05:
noted