zacharee/Bifrost #18: No changelog with the history tab

CiprianDinca9, 2021-07-02:
Samsung provides changelogs for firmwares. By typing model number notify update, you can get the changelogs. I know, it is Samsung mobile, doc.samsungmobile.com. now that we can no longer download from the tab, you can update the website.

zacharee, 2021-07-02:
Can you provide an example URL?

CiprianDinca9, 2021-07-03:
[]()https://doc.samsungmobile.com/SM-A202F/DBT/doc.html

CiprianDinca9, 2021-07-03:
I have commented on GitHub. You can see the link there. That provides lots
of info, especially made for such features. You need to know, I really like
the app and will donate as soon as I can!

zacharee, 2021-07-04:
Build 0.5.0 will have a changelog for both the retrieved firmware in the Download tab and for each item in the History tab.

closed 2021-07-04 by zacharee