zacharee/Bifrost #34: Strange issue on Arch Linux (latest, KDE)

TheAirBlow, 2021-09-03:
Just when I try to download latest firmware for SM-A207F region SER it opens a window that is not rendering (transparent in KDE)

When I close that window, the download ends. 
No logs found.

zacharee, 2021-09-04:
That should be the file picker. If I remember correctly, I've tested in KDE without issue. Make sure you have Dolphin installed.

TheAirBlow, 2021-09-04:
It is installed. Maybe some dependencies are missing? Any other app can
open this dialog.

zacharee, 2021-09-04:
I'm not sure. I'll test it out on a KDE machine to see what's going on. What OS and version are you using?

TheAirBlow, 2021-09-05:
```
                   -`                    theairblow@theairblow 
                  .o+`                   --------------------- 
                 `ooo/                   OS: Arch Linux x86_64 
                `+oooo:                  Host: 81LW Lenovo IdeaPad L340-15API 
               `+oooooo:                 Kernel: 5.13.13-arch1-1 
               -+oooooo+:                Uptime: 1 hour, 13 mins 
             `/:-:++oooo+:               Packages: 1607 (pacman), 15 (snap) 
            `/++++/+++++++:              Shell: zsh 5.8 
           `/++++++++++++++:             Resolution: 1920x1080 
          `/+++ooooooooooooo/`           DE: Plasma 5.22.4 
         ./ooosssso++osssssso+`          WM: KWin 
        .oossssso-````/ossssss+`         Theme: Breeze Dark [Plasma], Breeze [GTK2/3] 
       -osssssso.      :ssssssso.        Icons: Papirus-Bluegrey-Dark [Plasma], Papirus-Bluegrey-Dark [GTK2/3 
      :osssssss/        osssso+++.       Terminal: konsole 
     /ossssssss/        +ssssooo/-       Terminal Font: DejaVu Sans Mono for Powerline 10 
   `/ossssso+/:-        -:/+osssso+-     CPU: AMD Ryzen 3 3200U with Radeon Vega Mobile Gfx (4) @ 2.600GHz 
  `+sso+:-`                 `.-/+oso:    GPU: AMD ATI 03:00.0 Picasso 
 `++:.                           `-/+/   Memory: 1974MiB / 5809MiB 
 .`                                 `/
                                                                 
                                                                 
```

kujaw, 2022-01-12:
Same here, using Xfce:

`                  .o+`                 k@k
                 `ooo/                 OS: Arch Linux 
                `+oooo:                Kernel: x86_64 Linux 5.15.12-arch1-1
               `+oooooo:               Uptime: 2d 15h 51m
               -+oooooo+:              Packages: 1723
             `/:-:++oooo+:             Shell: bash 5.1.12
            `/++++/+++++++:            Resolution: 1920x1080
           `/++++++++++++++:           DE: Xfce4
          `/+++ooooooooooooo/`         WM: Xfwm4
         ./ooosssso++osssssso+`        WM Theme: Layan-light
        .oossssso-````/ossssss+`       GTK Theme: Layan-light [GTK2]
       -osssssso.      :ssssssso.      Icon Theme: Tela-manjaro
      :osssssss/        osssso+++.     Font: Lato 11
     /ossssssss/        +ssssooo/-     Disk: 754G / 1.2T (70%)
   `/ossssso+/:-        -:/+osssso+-   CPU: Intel Core i5-4340M @ 4x 3.6GHz [58.0°C]
  `+sso+:-`                 `.-/+oso:  GPU: Intel Corporation 4th Gen Core Processor Integrated Graphics Controller (rev 06)
 `++:.                           `-/+/ RAM: 9684MiB / 15914MiB
 .`                                 `/
`

kujaw, 2022-01-12:
Log:

```
$ ./Bifrost 
Generating nonce.
Nonce: cd95gvs5b7a7x0no
Auth: m1Ow1UGLHjfJW/9WB46q2vX5OpJPyoLZw5QgCt+qVog=

(Bifrost:28069): Gtk-WARNING **: 22:15:05.632: drawing failure for widget 'GtkLabel': error occurred in libfreetype

(Bifrost:28069): Gtk-WARNING **: 22:15:05.632: drawing failure for widget 'GtkBox': error occurred in libfreetype

(Bifrost:28069): Gtk-WARNING **: 22:15:05.632: drawing failure for widget 'GtkEventBox': error occurred in libfreetype

(Bifrost:28069): Gtk-WARNING **: 22:15:05.632: drawing failure for widget 'GtkRevealer': error occurred in libfreetype

(Bifrost:28069): Gtk-WARNING **: 22:15:05.632: drawing failure for widget 'GtkSidebarRow': error occurred in libfreetype

(Bifrost:28069): Gtk-WARNING **: 22:15:05.633: drawing failure for widget 'GtkListBox': error occurred in libfreetype

(Bifrost:28069): Gtk-WARNING **: 22:15:05.633: drawing failure for widget 'GtkViewport': error occurred in libfreetype

(Bifrost:28069): Gtk-WARNING **: 22:15:05.633: drawing failure for widget 'GtkPlacesSidebar': error occurred in libfreetype

(Bifrost:28069): Gtk-WARNING **: 22:15:05.635: drawing failure for widget 'GtkPaned': error occurred in libfreetype

(Bifrost:28069): Gtk-WARNING **: 22:15:05.637: drawing failure for widget 'GtkBox': error occurred in libfreetype

(Bifrost:28069): Gtk-WARNING **: 22:15:05.638: drawing failure for widget 'GtkFileChooserWidget': error occurred in libfreetype

(Bifrost:28069): Gtk-WARNING **: 22:15:05.638: drawing failure for widget 'GtkBox': error occurred in libfreetype

(Bifrost:28069): Gtk-WARNING **: 22:15:05.638: drawing failure for widget 'GtkFileChooserDialog': error occurred in libfreetype

(Bifrost:28069): Gtk-WARNING **: 22:15:05.650: drawing failure for widget 'GtkTreeView': error occurred in libfreetype

(Bifrost:28069): Gtk-WARNING **: 22:15:05.650: drawing failure for widget 'GtkScrolledWindow': error occurred in libfreetype

(Bifrost:28069): Gtk-WARNING **: 22:15:05.650: drawing failure for widget 'GtkBox': error occurred in libfreetype

(Bifrost:28069): Gtk-WARNING **: 22:15:05.650: drawing failure for widget 'GtkStack': error occurred in libfreetype

(Bifrost:28069): Gtk-WARNING **: 22:15:05.650: drawing failure for widget 'GtkBox': error occurred in libfreetype

(Bifrost:28069): Gtk-WARNING **: 22:15:05.650: drawing failure for widget 'GtkBox': error occurred in libfreetype

(Bifrost:28069): Gtk-WARNING **: 22:15:05.650: drawing failure for widget 'GtkPaned': error occurred in libfreetype

(Bifrost:28069): Gtk-WARNING **: 22:15:05.650: drawing failure for widget 'GtkBox': error occurred in libfreetype

(Bifrost:28069): Gtk-WARNING **: 22:15:05.650: drawing failure for widget 'GtkFileChooserWidget': error occurred in libfreetype

(Bifrost:28069): Gtk-WARNING **: 22:15:05.650: drawing failure for widget 'GtkBox': error occurred in libfreetype

(Bifrost:28069): Gtk-WARNING **: 22:15:05.650: drawing failure for widget 'GtkFileChooserDialog': error occurred in libfreetype

(Bifrost:28069): Gtk-WARNING **: 22:15:05.665: drawing failure for widget 'GtkFileChooserEntry': error occurred in libfreetype

(Bifrost:28069): Gtk-WARNING **: 22:15:05.665: drawing failure for widget 'GtkBox': error occurred in libfreetype

(Bifrost:28069): Gtk-WARNING **: 22:15:05.665: drawing failure for widget 'GtkHeaderBar': error occurred in libfreetype

(Bifrost:28069): Gtk-WARNING **: 22:15:05.665: drawing failure for widget 'GtkFileChooserDialog': error occurred in libfreetype

(Bifrost:28069): Gtk-WARNING **: 22:15:05.681: drawing failure for widget 'GtkFileChooserEntry': error occurred in libfreetype

(Bifrost:28069): Gtk-WARNING **: 22:15:05.681: drawing failure for widget 'GtkBox': error occurred in libfreetype

(Bifrost:28069): Gtk-WARNING **: 22:15:05.681: drawing failure for widget 'GtkHeaderBar': error occurred in libfreetype

(Bifrost:28069): Gtk-WARNING **: 22:15:05.681: drawing failure for widget 'GtkFileChooserDialog': error occurred in libfreetype

(Bifrost:28069): Gtk-WARNING **: 22:15:05.697: drawing failure for widget 'GtkFileChooserEntry': error occurred in libfreetype

(Bifrost:28069): Gtk-WARNING **: 22:15:05.698: drawing failure for widget 'GtkBox': error occurred in libfreetype

(Bifrost:28069): Gtk-WARNING **: 22:15:05.698: drawing failure for widget 'GtkHeaderBar': error occurred in libfreetype

(Bifrost:28069): Gtk-WARNING **: 22:15:05.698: drawing failure for widget 'GtkFileChooserDialog': error occurred in libfreetype

(Bifrost:28069): Gtk-WARNING **: 22:15:06.740: drawing failure for widget 'GtkTreeView': error occurred in libfreetype

(Bifrost:28069): Gtk-WARNING **: 22:15:06.740: drawing failure for widget 'GtkScrolledWindow': error occurred in libfreetype

(Bifrost:28069): Gtk-WARNING **: 22:15:06.740: drawing failure for widget 'GtkBox': error occurred in libfreetype

(Bifrost:28069): Gtk-WARNING **: 22:15:06.740: drawing failure for widget 'GtkStack': error occurred in libfreetype

(Bifrost:28069): Gtk-WARNING **: 22:15:06.740: drawing failure for widget 'GtkBox': error occurred in libfreetype

(Bifrost:28069): Gtk-WARNING **: 22:15:06.740: drawing failure for widget 'GtkBox': error occurred in libfreetype

(Bifrost:28069): Gtk-WARNING **: 22:15:06.740: drawing failure for widget 'GtkPaned': error occurred in libfreetype
.,.
```

zacharee, 2022-02-25:
1.0.4 uses a new version of Compose. Can you see if it works better?

Lawyno, 2022-03-19:
I am also having this issue with v1.0.5.
Log is almost the same like the one kujaw posted.

Operating System: Manjaro Linux
KDE Plasma Version: 5.24.3
KDE Frameworks Version: 5.91.0
Qt Version: 5.15.3
Kernel Version: 5.16.14-1-MANJARO (64-bit)
Graphics Platform: X11

bziur, 2022-04-07:
This could be related to freetype bugs in particular specyfic to perticular fonts and characters.
There seems to be a [thread related to this on debian bugtracker](https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=998064), and on [JetBrains issues](https://github.com/JetBrains/compose-jb/issues/1279)

zacharee, 2022-04-15:
I think Arch just has some issue with the native file picker. There are a bunch of similar JDK bugs that have supposedly been fixed, but even packaging with JDK18, when I open the file picker in my Arch VM, the entire JDK crashes.

[Release 1.0.5-exp5](https://github.com/zacharee/SamloaderKotlin/releases/tag/1.0.5-exp5) uses the Swing file picker instead of the native one. It doesn't look as nice, but it should at least function.

zacharee, 2022-05-16:
Closing as fixed for now.

closed 2022-05-16 by zacharee