zacharee/Bifrost #46: Trying to Download Firmware Results in a weird Error(?)

adikpb, 2021-12-27:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

zacharee, 2021-12-27:
Samsung isn't hosting the firmware for that device and CSC. Try XFA.

adikpb, 2021-12-28:
Ah I see. Btw did you mean XDA? Thank you.

zacharee, 2021-12-28:
No, XFA is the South Africa CSC, which does have firmware available.

Zachary Wander

adikpb, 2021-12-28:
Ah thank you.

closed 2021-12-28 by adikpb