zacharee/Bifrost #54: ClientRequestException, 403 Forbidden, Access denied

Trikenstein, 2022-02-16:
BiFrost 1.0.3, Linux, Ubuntu 20.04.3

- Model: **SM-P610**
- Region: **XAC**

```
io.ktor.client.features.ClientRequestException: Client request(https://fota-cloud-dn.ospserver.net:443/firmware/XAC/SM-P160/versio
n.xml) invalid: 403 Forbidden. Text: "<?xml version="1.0" encoding="UTF-8"?>                                                      
<Error><Code>AccessDenied</Code><Message>Access Denied</Message><RequestId>1KJ2BJXQ4DJR18KC</RequestId><HostId>naeVGuvPLdw5G8CNI5B
03c6D8+FIhE5jMfLiu03bvepLjGAQ7XgjVqT8msiJf3TR6bqCkRq0Hs4=</HostId></Error>"                                                       
        at io.ktor.client.features.DefaultResponseValidationKt$addDefaultResponseValidation$1$1.invokeSuspend(DefaultResponseValid
ation.kt:47)                                                                                                                      
        at io.ktor.client.features.DefaultResponseValidationKt$addDefaultResponseValidation$1$1.invoke(DefaultResponseValidation.k
t)                                                                                                                                
        at io.ktor.client.features.DefaultResponseValidationKt$addDefaultResponseValidation$1$1.invoke(DefaultResponseValidation.k
t)                                                                                                                                
        at io.ktor.client.features.HttpCallValidator.validateResponse(HttpCallValidator.kt:54)                                    
        at io.ktor.client.features.HttpCallValidator.access$validateResponse(HttpCallValidator.kt:33)                             
        at io.ktor.client.features.HttpCallValidator$Companion$install$3.invokeSuspend(HttpCallValidator.kt:133)                  
        at io.ktor.client.features.HttpCallValidator$Companion$install$3.invoke(HttpCallValidator.kt)                             
        at io.ktor.client.features.HttpCallValidator$Companion$install$3.invoke(HttpCallValidator.kt)                             
        at io.ktor.client.features.HttpSend$Feature$install$1.invokeSuspend(HttpSend.kt:96)                                       
        at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)                                 
        at io.ktor.util.pipeline.SuspendFunctionGun.resumeRootWith(SuspendFunctionGun.kt:191)                                     
        at io.ktor.util.pipeline.SuspendFunctionGun.loop(SuspendFunctionGun.kt:147)
        at io.ktor.util.pipeline.SuspendFunctionGun.access$loop(SuspendFunctionGun.kt:15)
        at io.ktor.util.pipeline.SuspendFunctionGun$continuation$1.resumeWith(SuspendFunctionGun.kt:93)
        at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:46)
        at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:106)
        at java.desktop/java.awt.event.InvocationEvent.dispatch(Unknown Source)
        at java.desktop/java.awt.EventQueue.dispatchEventImpl(Unknown Source)
        at java.desktop/java.awt.EventQueue$4.run(Unknown Source) 
        at java.desktop/java.awt.EventQueue$4.run(Unknown Source) 
        at java.base/java.security.AccessController.doPrivileged(Unknown Source)
        at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
        at java.desktop/java.awt.EventQueue.dispatchEvent(Unknown Source)
        at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
        at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
        at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
        at java.desktop/java.awt.EventDispatchThread.pumpEvents(Unknown Source)
        at java.desktop/java.awt.EventDispatchThread.pumpEvents(Unknown Source)
        at java.desktop/java.awt.EventDispatchThread.run(Unknown Source)

Exception in thread "AWT-EventQueue-0" io.ktor.client.features.ClientRequestException: Client request(https://fota-cloud-dn.ospserver.net:443/firmware/XAC/SM-P160/version.xml) invalid: 403 Forbidden. Text: "<?xml version="1.0" encoding="UTF-8"?>
<Error><Code>AccessDenied</Code><Message>Access Denied</Message><RequestId>1KJ2BJXQ4DJR18KC</RequestId><HostId>naeVGuvPLdw5G8CNI5B03c6D8+FIhE5jMfLiu03bvepLjGAQ7XgjVqT8msiJf3TR6bqCkRq0Hs4=</HostId></Error>"
        at io.ktor.client.features.DefaultResponseValidationKt$addDefaultResponseValidation$1$1.invokeSuspend(DefaultResponseValidation.kt:47)
        at io.ktor.client.features.DefaultResponseValidationKt$addDefaultResponseValidation$1$1.invoke(DefaultResponseValidation.kt)
        at io.ktor.client.features.DefaultResponseValidationKt$addDefaultResponseValidation$1$1.invoke(DefaultResponseValidation.kt)
        at io.ktor.client.features.HttpCallValidator.validateResponse(HttpCallValidator.kt:54)
        at io.ktor.client.features.HttpCallValidator.access$validateResponse(HttpCallValidator.kt:33)
        at io.ktor.client.features.HttpCallValidator$Companion$install$3.invokeSuspend(HttpCallValidator.kt:133)
        at io.ktor.client.features.HttpCallValidator$Companion$install$3.invoke(HttpCallValidator.kt)
        at io.ktor.client.features.HttpCallValidator$Companion$install$3.invoke(HttpCallValidator.kt)
        at io.ktor.client.features.HttpSend$Feature$install$1.invokeSuspend(HttpSend.kt:96)
        at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
        at io.ktor.util.pipeline.SuspendFunctionGun.resumeRootWith(SuspendFunctionGun.kt:191)
        at io.ktor.util.pipeline.SuspendFunctionGun.loop(SuspendFunctionGun.kt:147)
        at io.ktor.util.pipeline.SuspendFunctionGun.access$loop(SuspendFunctionGun.kt:15)
        at io.ktor.util.pipeline.SuspendFunctionGun$continuation$1.resumeWith(SuspendFunctionGun.kt:93)
        at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:46)
        at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:106)
        at java.desktop/java.awt.event.InvocationEvent.dispatch(Unknown Source)
        at java.desktop/java.awt.EventQueue.dispatchEventImpl(Unknown Source)
        at java.desktop/java.awt.EventQueue$4.run(Unknown Source)
        at java.desktop/java.awt.EventQueue$4.run(Unknown Source)
        at java.base/java.security.AccessController.doPrivileged(Unknown Source)
        at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
        at java.desktop/java.awt.EventQueue.dispatchEvent(Unknown Source)
        at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
        at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
        at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
        at java.desktop/java.awt.EventDispatchThread.pumpEvents(Unknown Source)
        at java.desktop/java.awt.EventDispatchThread.pumpEvents(Unknown Source)
        at java.desktop/java.awt.EventDispatchThread.run(Unknown Source)
```

closed 2022-02-25 by zacharee

Trikenstein, 2022-02-25:
Can you please give a short comment about closing this issue?

zacharee, 2022-02-25:
The crash is fixed in 1.0.4.