zacharee/Bifrost #57: 403 permission issue on fota-cloud-dn.ospserver.net:443

IHateNameStealers, 2022-03-21:
Getting the following error at this moment. (due to a permission issue on: https://fota-cloud-dn.ospserver.net:443/firmware/TNL/SM-F96B/version.xml](https://fota-cloud-dn.ospserver.net/firmware/TNL/SM-F96B/version.xml

Error / warning shown within BitFrost.

> lidation.kt)
at io.ktor.client.features.DefaultResponseValidationKt$addDefaultResponseValidation$1$1.invoke(DefaultResponseValidation.kt)
at io.ktor.client.features.HttpCallValidator.validateResponse(HttpCallValidator.kt:54)
at io.ktor.client.features.HttpCallValidator.access$validateResponse(HttpCallValidator.kt:33)
at io.ktor.client.features.HttpCallValidator$Companion$install$3.invokeSuspend(HttpCallValidator.kt:133)
at io.ktor.client.features.HttpCallValidator$Companion$install$3.invoke(HttpCallValidator.kt)
at io.ktor.client.features.HttpCallValidator$Companion$install$3.invoke(HttpCallValidator.kt)
at io.ktor.client.features.HttpSend$Feature$install$1.invokeSuspend(HttpSend.kt:96)
at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
at io.ktor.util.pipeline.SuspendFunctionGun.resumeRootWith(SuspendFunctionGun.kt:191)
at io.ktor.util.pipeline.SuspendFunctionGun.loop(SuspendFunctionGun.kt:147)
at io.ktor.util.pipeline.SuspendFunctionGun.access$loop(SuspendFunctionGun.kt:15)
at io.ktor.util.pipeline.SuspendFunctionGun$continuation$1.resumeWith(SuspendFunctionGun.kt:93)
at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:46)
at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:106)
at java.desktop/java.awt.event.InvocationEvent.dispatch(Unknown Source)
at java.desktop/java.awt.EventQueue.dispatchEventImpl(Unknown Source)
at java.desktop/java.awt.EventQueue$4.run(Unknown Source)
at java.desktop/java.awt.EventQueue$4.run(Unknown Source)
at java.base/java.security.AccessController.doPrivileged(Unknown Source)
at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
at java.desktop/java.awt.EventQueue.dispatchEvent(Unknown Source)
at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
at java.desktop/java.awt.EventDispatchThread.pumpEvents(Unknown Source)
at java.desktop/java.awt.EventDispatchThread.pumpEvents(Unknown Source)
at java.desktop/java.awt.EventDispatchThread.run(Unknown Source)

Leo17032009, 2022-03-22:
https://github.com/zacharee/SamloaderKotlin#bifrost-is-returning-error-403-when-checking-for-updates

IHateNameStealers, 2022-03-22:
As mentioned above. I've tried several (euro) region/CSC, and all are giving me a 403 issue.

Leo17032009, 2022-03-22:
Make sure the model and region are correct.

Leo17032009, 2022-03-23:
I don't know SM-F96B. Please open Settings-About phone-Model.

IHateNameStealers, 2022-03-23:
Model and region are correct.
I used Bifrost some days ago, and all was working fine till then.
The  SM-F96B is the Z-Fold3 from Samsung :-)
The region I'm in is the Netherlands.

Leo17032009, 2022-03-24:
Thanke you. But I can't help you any more. :-(

Leo17032009, 2022-03-24:
Yes. I can download Samsung firmware for SM-A715F/SEK.

zacharee, 2022-04-15:
The Fold 3's model number is SM-F926B, not SM-F96B.

closed 2022-04-15 by zacharee