zacharee/Bifrost #58: More Info: connection reset

LeoAdavirat, 2022-03-26:
I tried to download a firmware and got into this issue, looking forward to a fix. Thank you in advance
[image]

zacharee, 2022-04-15:
This is an issue with your internet connection.

Leo17032009, 2022-04-15:
> .
No, I have the same problem.[image]

zacharee, 2022-04-15:
> > .
> > No, I have the same problem.[image]

Your error is 403 and it's because you put in XVV instead of XXV

closed 2022-04-15 by zacharee