zacharee/Bifrost #62: All macOS version of the App are damaged after extracting and moving.

TheFirstFarmerBob, 2022-05-09:
I have downloaded several copies of the Bifrost_macOS.zip. After unzipping and opening the .dmg, no matter where the file is executed from or moved to, upon trying to open the app I get an error saying that the file is damaged and to throw it in the trash.

I have read several other areas where they discuss their macOS file/app becoming corrupt or losing "permissions" when being zipped. So when I saw this in a DMG I thought, genius. Now just might work. Sorry, for me it didn't.

If this issue has been presented already please disregard my post.

Leo17032009, 2022-05-09:
Check with the command `sudo xattr -cr /Applications/Samsung\ Firmware\ Downloader.app` #42

zacharee, 2022-05-12:
Please see the instructions in the README: https://github.com/zacharee/SamloaderKotlin#macos

closed 2022-05-12 by zacharee

TheFirstFarmerBob, 2022-05-13:
Thank you for your reply. I did read your READ ME first thing, always do, and prior to downloading the file. I am familiar with what it discusses and have gone through the actions to remedy a Security Block. What I experience was not the Security Block. Although, it did have to go through it with the latest .dmg. The contained file was deemed "Damaged" not "Security Locked". I just DLd the .DMG again and got the same results. Then tried a second time, this time I was able to open the file.

I did notice that the first several times when the file failed, the DMG would not eject. So prior to trying to open the file, I closed the DMG, which it did easily. It looks as though the copy process stalled. I've seen it before. Thus the file failing to open and being deemed "damaged".

zacharee, 2022-05-13:
macOS also calls security blocked files "damaged".

TheFirstFarmerBob, 2022-05-13:
Although I have never seen that, it may very well be. I did not get the error panels shown in the READ ME. I got a single small panel with the "Damaged" wording and the ever loved, "OK" button. I went to the Security Panel to see if the file was listed as they also do sometimes. It was not. So there was no out as describe. 

But as I said above I was able to open the file recently in a second try once I verified that the "copy" process was complete or there was no complications and the DMG ejected cleanly. Then the app opened right up. Back in the day when I use to use FTP and whatnot to move files, I'd have this problem every so often. Although Zipped or so, the Internet still has to mess with them sometimes. 

I so look forward to using it to recover a "Samsung" trashed tablet. Just need to hunt down all the info needed that I use to know by heart. It's been a while.

Thanks again.