zacharee/Bifrost #67: "Working..." notification not dismissing when closing the app.

Leo17032009, 2022-06-04:
[image]

zacharee, 2022-06-17:
Have you removed the app from Recents?

Leo17032009, 2022-06-18:
> Have you removed the app from Recents?

How do I do this?

zacharee, 2022-06-18:
https://user-images.githubusercontent.com/9020352/174441994-51e22d27-b0b5-492b-9aad-aaa5ae105e08.mp4

Leo17032009, 2022-06-18:
https://drive.google.com/file/d/1MKOGYShn5Ys2F6mHqhAOUIJp1hfyQbAQ/view?usp=drivesdk

languagemaniac, 2023-01-03:
I am also having this issue. It successfully downloaded and decrypted a firmware but the notification got stucked saying "working..."and "decrypting 99.98%"

I had to force close the app from the settings. As closing the program did nothing.

Opening the app again causes the same notification to appear (only it doesn't display "decrypting" info) , but closing the program does make the notification disappear.

Other than that it works great.

I hope this information can help you fix the issue.

closed 2023-11-11 by zacharee

jmichael2497, 2025-07-26:
still happening on stock SM-T500 with final A12 running v1.20.2 android app seems to show successful download and decrypt, but notification was stuck saying something like decrypt in progress 99.2% so closed app to clear that.