zacharee/Bifrost #84: CRC Verify Failed

thewolfprod, 2022-12-25:
Happens everytime I'm trying to download firmware on 400 - 500 mb downloaded.
[image]

zacharee, 2023-01-11:
Try a different region code. Samsung may not be properly serving the firmware for that region.

closed 2023-02-15 by zacharee

namrabtw, 2023-09-23:
[image]

I always get the same thing and changing region isn't helping.

rscm, 2023-10-11:
Same here

OS: Linux Fedora Silverblue 38

Changed between all csc of my country, same result