zacharee/Bifrost #86: Can't build desktop linux with latest android studio canary

matan7890, 2022-12-27:
Hey zach,
I'm trying to build the linux desktop version using latest android studio canary (2022.2.1 Canary 10) by following the README.md file.

The problem is that in the readme it is written to build the `createDistributable` task in the gradle view, but there is no such task for desktop, only the task `jvmTest` exists. Tried on both the current master and on version 1.12.0 tag.
Of course building jvmTest achieves nothing.

Any ideas what am I missing? Maybe the README file is not up to date?

Thanks in advanced!

parkerlreed, 2022-12-28:
The releases are kept up to date (last 4 days ago). 

Are you trying to modify the source?

msaiko239, 2022-12-28:
I was facing the same issue, under Settings > Experimental you need to un-check 
`Only include test task in the Gradle test list generated during Gradle Sync`

matan7890, 2023-01-04:
@parkerlreed I prefer to build them by myself if possible.

@msaiko239 Thanks I'll check this out. Is it possible to update the build manual for dumb users who does not use Android Studio regularly like me?

closed 2023-02-13 by zacharee