zacharee/Bifrost #89: Crashes when trying to enter something I'm the model number field.

hyperio546, 2023-02-06:
Using an Android phone. Samsung galaxy A03 running one UI 5 core rooted.

zeca22, 2023-02-06:
same here
samsung s22+   A13
i cant do anything from app, always  crashing
[Log_2023-02-06_05-42-08.txt](https://github.com/zacharee/SamloaderKotlin/files/10660064/Log_2023-02-06_05-42-08.txt)

hyperio546, 2023-02-06:
> same here
> samsung s22+   A13
> i cant do anything from app, always  crashing
> [Log_2023-02-06_05-42-08.txt](https://github.com/zacharee/SamloaderKotlin/files/10660064/Log_2023-02-06_05-42-08.txt)

This version seems pretty much broken. Downgrading works.

zacharee, 2023-02-11:
Should be fixed with https://github.com/zacharee/SamloaderKotlin/releases/tag/1.13.1

closed 2023-02-11 by zacharee

coderx546, 2023-02-12:
> Should be fixed with https://github.com/zacharee/SamloaderKotlin/releases/tag/1.13.1

Sure thanks.