zacharee/Bifrost #90: Windows and Android versions are not working properly.

Star-Seven, 2023-02-15:
[CUsersStarDesktopBifrostBifrost.exe.txt](https://github.com/zacharee/SamloaderKotlin/files/10740384/CUsersStarDesktopBifrostBifrost.exe.txt)
[data_app_crash@1676445870170.txt](https://github.com/zacharee/SamloaderKotlin/files/10740385/data_app_crash%401676445870170.txt)

zacharee, 2023-02-15:
Are you connected to the internet?

Star-Seven, 2023-02-16:
> 
Yes, there is no problem with my network. The old version of Bifrost (v1.0.11) can also be used normally. Versions after v1.0.11 will not work properly!
[image]
[image]

zacharee, 2023-02-16:
I can't reproduce this issue. The part where it's crashing is attempting to connect to `raw.githubusercontent.com`. Do you have that domain blocked?

Star-Seven, 2023-02-16:
No blocking was done. This should be my network provider's problem. But in the same case, I can use the old version v1.0.11 normally.

zacharee, 2023-02-18:
1.14.0 should no longer crash.

Star-Seven, 2023-02-18:
Ok, thank you very much. I will test it!

Star-Seven, 2023-02-18:
> 1.14.0 should no longer crash.

V1.14.0 has been tested. Windows and Android versions can work normally. Thank you!

closed 2023-02-18 by zacharee