zacharee/Bifrost #94: Bad Return status: 400 of Galaxy Note 4 SM-N910C XEO

Unknown78, 2023-03-22:
I have run as Admin
I have the latest Bifrost 1.14.1
I have the latest Eclipse Temurin JDK with Hotspot 19.0.2+7

`N910CXXU2DVH2/N910COXA2DVH2/N910CXXU2DVH2/N910CXXU2DVH2`

[image]

P.S. This is on a machine with Intel HD Graphics 5300 as the only GPU.

Unknown78, 2023-03-22:
This problem is similar with samfirm.js that it can't download the firmware properly.
I have opened the issue there but received no reply yet:
[TypeError: Cannot read properties of undefined (reading 'BINARY_BYTE_SIZE')](https://github.com/jesec/samfirm.js/issues/5)

Unknown78, 2023-03-31:
@zacharee Could you please help me?

zacharee, 2023-03-31:
This is on Samsung's end. Try a different CSC.

Unknown78, 2023-03-31:
@zacharee 

What do you mean by Samsung's end?
This is the latest firmware of that CSC, the issue should be different than https://github.com/zacharee/SamloaderKotlin/issues/10

zacharee, 2023-03-31:
Samsung sometimes doesn't serve firmware for older devices on certain CSCs and will return error 400. There isn't anything you can do about it except try a different CSC.

Unknown78, 2023-03-31:
@zacharee 
But it is stated that the release date is 2022-08-19
https://doc.samsungmobile.com/SM-N910C/XEO/doc.html

The device is old, but the firmware is still fresh, how could it be that we can't download it?

zacharee, 2023-03-31:
You're going to have to ask Samsung's internal engineering department.

Unknown78, 2023-04-01:
@zacharee How to ask them? Do you have the access to contact them? Will they really give you for free so easily? Have you ever tried it?

closed 2023-11-11 by zacharee