zacharee/Bifrost #96: Issues when building the app from sources

BlackMesa123, 2023-04-24:
I'm currently having some issues building the app from sources with the provided guide in README, I'm using the latest Android Studio Canary build. By cloning the repo, opening the project in AS, running the `:android:build` task throws this error:
```
> Task :android:mergeDexRelease FAILED
AGPBI: {"kind":"error","text":"Type tk.zwander.samsungfirmwaredownloader.BuildConfig is defined multiple times: /Users/salvo/Desktop/Android/Sorgenti/Apps/SamloaderKotlin/android/build/intermediates/project_dex_archive/release/out/tk/zwander/samsungfirmwaredownloader/BuildConfig.dex, /Users/salvo/Desktop/Android/Sorgenti/Apps/SamloaderKotlin/common/build/.transforms/eb9b6aa470043c7f4cb6f65ed6a9784e/transformed/release/tk/zwander/samsungfirmwaredownloader/BuildConfig.dex","sources":[{"file":"/Users/salvo/Desktop/Android/Sorgenti/Apps/SamloaderKotlin/android/build/intermediates/project_dex_archive/release/out/tk/zwander/samsungfirmwaredownloader/BuildConfig.dex"}],"tool":"D8"}
com.android.builder.dexing.DexArchiveMergerException: Error while merging dex archives: 
Learn how to resolve the issue at https://developer.android.com/studio/build/dependencies#duplicate_classes.
Type tk.zwander.samsungfirmwaredownloader.BuildConfig is defined multiple times: /Users/salvo/Desktop/Android/Sorgenti/Apps/SamloaderKotlin/android/build/intermediates/project_dex_archive/release/out/tk/zwander/samsungfirmwaredownloader/BuildConfig.dex, /Users/salvo/Desktop/Android/Sorgenti/Apps/SamloaderKotlin/common/build/.transforms/eb9b6aa470043c7f4cb6f65ed6a9784e/transformed/release/tk/zwander/samsungfirmwaredownloader/BuildConfig.dex
	at com.android.builder.dexing.D8DexArchiveMerger.getExceptionToRethrow(D8DexArchiveMerger.java:151)
	at com.android.builder.dexing.D8DexArchiveMerger.mergeDexArchives(D8DexArchiveMerger.java:138)
	at com.android.build.gradle.internal.tasks.DexMergingWorkAction.merge(DexMergingTask.kt:858)
	at com.android.build.gradle.internal.tasks.DexMergingWorkAction.run(DexMergingTask.kt:804)
	at com.android.build.gradle.internal.profile.ProfileAwareWorkAction.execute(ProfileAwareWorkAction.kt:74)
	at org.gradle.workers.internal.DefaultWorkerServer.execute(DefaultWorkerServer.java:63)
	at org.gradle.workers.internal.NoIsolationWorkerFactory$1$1.create(NoIsolationWorkerFactory.java:66)
	at org.gradle.workers.internal.NoIsolationWorkerFactory$1$1.create(NoIsolationWorkerFactory.java:62)
	at org.gradle.internal.classloader.ClassLoaderUtils.executeInClassloader(ClassLoaderUtils.java:100)
	at org.gradle.workers.internal.NoIsolationWorkerFactory$1.lambda$execute$0(NoIsolationWorkerFactory.java:62)
	at org.gradle.workers.internal.AbstractWorker$1.call(AbstractWorker.java:44)
	at org.gradle.workers.internal.AbstractWorker$1.call(AbstractWorker.java:41)
	at org.gradle.internal.operations.DefaultBuildOperationRunner$CallableBuildOperationWorker.execute(DefaultBuildOperationRunner.java:204)
	at org.gradle.internal.operations.DefaultBuildOperationRunner$CallableBuildOperationWorker.execute(DefaultBuildOperationRunner.java:199)
	at org.gradle.internal.operations.DefaultBuildOperationRunner$2.execute(DefaultBuildOperationRunner.java:66)
	at org.gradle.internal.operations.DefaultBuildOperationRunner$2.execute(DefaultBuildOperationRunner.java:59)
	at org.gradle.internal.operations.DefaultBuildOperationRunner.execute(DefaultBuildOperationRunner.java:157)
	at org.gradle.internal.operations.DefaultBuildOperationRunner.execute(DefaultBuildOperationRunner.java:59)
	at org.gradle.internal.operations.DefaultBuildOperationRunner.call(DefaultBuildOperationRunner.java:53)
	at org.gradle.internal.operations.DefaultBuildOperationExecutor.call(DefaultBuildOperationExecutor.java:73)
	at org.gradle.workers.internal.AbstractWorker.executeWrappedInBuildOperation(AbstractWorker.java:41)
	at org.gradle.workers.internal.NoIsolationWorkerFactory$1.execute(NoIsolationWorkerFactory.java:59)
	at org.gradle.workers.internal.DefaultWorkerExecutor.lambda$submitWork$0(DefaultWorkerExecutor.java:169)
	at java.base/java.util.concurrent.FutureTask.run(Unknown Source)
	at org.gradle.internal.work.DefaultConditionalExecutionQueue$ExecutionRunner.runExecution(DefaultConditionalExecutionQueue.java:187)
	at org.gradle.internal.work.DefaultConditionalExecutionQueue$ExecutionRunner.access$700(DefaultConditionalExecutionQueue.java:120)
	at org.gradle.internal.work.DefaultConditionalExecutionQueue$ExecutionRunner$1.run(DefaultConditionalExecutionQueue.java:162)
	at org.gradle.internal.Factories$1.create(Factories.java:31)
	at org.gradle.internal.work.DefaultWorkerLeaseService.withLocks(DefaultWorkerLeaseService.java:249)
	at org.gradle.internal.work.DefaultWorkerLeaseService.runAsWorkerThread(DefaultWorkerLeaseService.java:109)
	at org.gradle.internal.work.DefaultWorkerLeaseService.runAsWorkerThread(DefaultWorkerLeaseService.java:114)
	at org.gradle.internal.work.DefaultConditionalExecutionQueue$ExecutionRunner.runBatch(DefaultConditionalExecutionQueue.java:157)
	at org.gradle.internal.work.DefaultConditionalExecutionQueue$ExecutionRunner.run(DefaultConditionalExecutionQueue.java:126)
	at java.base/java.util.concurrent.Executors$RunnableAdapter.call(Unknown Source)
	at java.base/java.util.concurrent.FutureTask.run(Unknown Source)
	at org.gradle.internal.concurrent.ExecutorPolicy$CatchAndRecordFailures.onExecute(ExecutorPolicy.java:64)
	at org.gradle.internal.concurrent.ManagedExecutorImpl$1.run(ManagedExecutorImpl.java:49)
	at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
	at java.base/java.lang.Thread.run(Unknown Source)
Caused by: com.android.tools.r8.CompilationFailedException: Compilation failed to complete, origin: /Users/salvo/Desktop/Android/Sorgenti/Apps/SamloaderKotlin/android/build/intermediates/project_dex_archive/release/out/tk/zwander/samsungfirmwaredownloader/BuildConfig.dex
	at Version.fakeStackEntry(Version_4.0.52.java:0)
	at com.android.tools.r8.internal.vk.a(R8_4.0.52_5a340ca2823c7e792fe09805c75f749b9d398d230bc0518bb54ae9b6b50addbe:75)
	at com.android.tools.r8.internal.vk.a(R8_4.0.52_5a340ca2823c7e792fe09805c75f749b9d398d230bc0518bb54ae9b6b50addbe:28)
	at com.android.tools.r8.internal.vk.a(R8_4.0.52_5a340ca2823c7e792fe09805c75f749b9d398d230bc0518bb54ae9b6b50addbe:27)
	at com.android.tools.r8.internal.vk.b(R8_4.0.52_5a340ca2823c7e792fe09805c75f749b9d398d230bc0518bb54ae9b6b50addbe:2)
	at com.android.tools.r8.D8.run(R8_4.0.52_5a340ca2823c7e792fe09805c75f749b9d398d230bc0518bb54ae9b6b50addbe:11)
	at com.android.builder.dexing.D8DexArchiveMerger.mergeDexArchives(D8DexArchiveMerger.java:136)
	... 38 more
Caused by: com.android.tools.r8.internal.g: Type tk.zwander.samsungfirmwaredownloader.BuildConfig is defined multiple times: /Users/salvo/Desktop/Android/Sorgenti/Apps/SamloaderKotlin/android/build/intermediates/project_dex_archive/release/out/tk/zwander/samsungfirmwaredownloader/BuildConfig.dex, /Users/salvo/Desktop/Android/Sorgenti/Apps/SamloaderKotlin/common/build/.transforms/eb9b6aa470043c7f4cb6f65ed6a9784e/transformed/release/tk/zwander/samsungfirmwaredownloader/BuildConfig.dex
	at com.android.tools.r8.internal.GV.a(R8_4.0.52_5a340ca2823c7e792fe09805c75f749b9d398d230bc0518bb54ae9b6b50addbe:14)
	at com.android.tools.r8.internal.GV.a(R8_4.0.52_5a340ca2823c7e792fe09805c75f749b9d398d230bc0518bb54ae9b6b50addbe:22)
	at com.android.tools.r8.internal.rP.a(R8_4.0.52_5a340ca2823c7e792fe09805c75f749b9d398d230bc0518bb54ae9b6b50addbe:44)
	at com.android.tools.r8.internal.rP.a(R8_4.0.52_5a340ca2823c7e792fe09805c75f749b9d398d230bc0518bb54ae9b6b50addbe:10)
	at java.base/java.util.concurrent.ConcurrentHashMap.merge(Unknown Source)
	at com.android.tools.r8.internal.rP.a(R8_4.0.52_5a340ca2823c7e792fe09805c75f749b9d398d230bc0518bb54ae9b6b50addbe:6)
	at com.android.tools.r8.graph.F2$a.e(R8_4.0.52_5a340ca2823c7e792fe09805c75f749b9d398d230bc0518bb54ae9b6b50addbe:7)
	at com.android.tools.r8.dex.b.a(R8_4.0.52_5a340ca2823c7e792fe09805c75f749b9d398d230bc0518bb54ae9b6b50addbe:84)
	at com.android.tools.r8.dex.b.a(R8_4.0.52_5a340ca2823c7e792fe09805c75f749b9d398d230bc0518bb54ae9b6b50addbe:23)
	at com.android.tools.r8.dex.b.a(R8_4.0.52_5a340ca2823c7e792fe09805c75f749b9d398d230bc0518bb54ae9b6b50addbe:22)
	at com.android.tools.r8.D8.a(R8_4.0.52_5a340ca2823c7e792fe09805c75f749b9d398d230bc0518bb54ae9b6b50addbe:50)
[... cut ...]
                                                                                                    	at androidx.compose.material3.IconButtonKt.IconButton(IconButton.kt:100)
                                                                                                    	at tk.zwander.commonCompose.view.components.FooterViewKt$FooterView$1$1$1$1$2.invoke(FooterView.kt:93)
                                                                                                    	at tk.zwander.commonCompose.view.components.FooterViewKt$FooterView$1$1$1$1$2.invoke(FooterView.kt:92)
```

Anything wrong in my side?

BlackMesa123, 2023-04-28:
The same issue happens when trying to build the app via CLI: https://github.com/BlackMesa123/SamloaderKotlin/actions/runs/4832737833/jobs/8611891261

zacharee, 2023-04-30:
I think this should be fixed in the latest commit.

BlackMesa123, 2023-04-30:
I can confirm commit 64fc960a3c7d0eb491bd465db564a7eeebed5b2a fixes the issue. Thank you.

closed 2023-04-30 by BlackMesa123