zacharee/Bifrost #105: Text Typed Into Entries Is Scrambled.

ethical-haquer, 2023-08-30:
When I type text into any entry, it becomes slightly scrambled. For example, "SM-A525F" turns into "S-A525FM" , "SM-A52F5", or "SM-A55F2" (these are all real examples). Another interesting example is me sliding my fingers over "QWERTYUIOP", which, although it works fine here, turns into "QWERYUIOPT", "WERTYUIOPQ", or "QERTYUIOPW" in Bifrost (also real examples). I've included a screen recording so you can _see_ what I'm talking about. It's not a big deal, but it _is_ annoying. :smile: TIA.

https://github.com/zacharee/SamloaderKotlin/assets/141518185/9830a9db-551c-457e-86b2-97a9dadde4e6

parkerlreed, 2023-09-20:
Yeah I've noticed this only happens on the Linux builds for me as well

ethical-haquer, 2023-09-24:
> Yeah I've noticed this only happens on the Linux builds for me as well

Yeah, this was on on Debian.

zacharee, 2023-11-10:
Compose Multiplatform sometimes has issues with text tracking on certain platforms. It might work better in 1.14.3, since I've updated the version of Compose used.

zacharee, 2024-01-15:
This shouldn't be an issue at all anymore in 1.16.14, which is currently building. Let me know if it still happens.

ethical-haquer, 2024-01-16:
> This shouldn't be an issue at all anymore in 1.16.14, which is currently building. Let me know if it still happens.

:+1:
I'm getting a fatal error using the v1.16.14 arm64 .deb package however:
```
ethical-haquer@raspberrypi:~/Apps $ bifrost
Jan 15, 2024 4:08:44 PM java.util.prefs.FileSystemPreferences$1 run
INFO: Created user preferences directory.
Jan 15, 2024 4:08:44 PM java.util.prefs.FileSystemPreferences$6 run
WARNING: Prefs file removed in background /home/ethical-haquer/.java/.userPrefs/prefs.xml
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  SIGSEGV (0xb) at pc=0x0000000000000000, pid=34792, tid=34825
#
# JRE version: OpenJDK Runtime Environment (18.0.2.1+1) (build 18.0.2.1+1-1)
# Java VM: OpenJDK 64-Bit Server VM (18.0.2.1+1-1, mixed mode, tiered, compressed oops, compressed class ptrs, g1 gc, linux-aarch64)
# Problematic frame:
# C  0x0000000000000000
#
# No core dump will be written. Core dumps have been disabled. To enable core dumping, try "ulimit -c unlimited" before starting Java again
#
# An error report file with more information is saved as:
# /home/ethical-haquer/Apps/hs_err_pid34792.log
#
# If you would like to submit a bug report, please visit:
#   https://bugreport.java.com/bugreport/crash.jsp
# The crash happened outside the Java Virtual Machine in native code.
# See problematic frame for where to report the bug.
#
Aborted
```
Here is the log:
[hs_err_pid34792.log](https://github.com/zacharee/SamloaderKotlin/files/13943960/hs_err_pid34792.log)
TIA.

zacharee, 2024-01-16:
Does 1.16.13 work?

ethical-haquer, 2024-01-16:
> Does 1.16.13 work?

I'll check right now...

ethical-haquer, 2024-01-16:
I'm getting the same error using v1.16.13 and v1.16.10. Should I try out some other versions to troubleshoot?

zacharee, 2024-01-16:
I can't reproduce on plain Ubuntu 22.04. Are you using Wayland with an NVIDIA card?

ethical-haquer, 2024-01-16:
> I can't reproduce on plain Ubuntu 22.04. Are you using Wayland with an NVIDIA card?
> 
> I tried both Wayland and Xorg.

I'm using Raspberry Pi OS on a Pi 5, the windowing system is wayland:
```
ethical-haquer@raspberrypi:~ $ echo $XDG_SESSION_TYPE
wayland
```

zacharee, 2024-01-16:
The closest I can get is Debian 12.4 running LXDE on X11 and lightdm and I can't reproduce. GNOME 40 and Plasma 5 on Wayland also work fine for me. 

The "problematic frame" in the crash log being "0" is also really strange. I think this is beyond what I can debug.

ethical-haquer, 2024-01-16:
> The closest I can get is Debian 12.4 running LXDE on X11 and lightdm and I can't reproduce. GNOME 40 and Plasma 5 on Wayland also work fine for me.
> 
> The "problematic frame" in the crash log being "0" is also really strange. I think this is beyond what I can debug.

Well, running `SKIKO_RENDER_API=DIRECT_SOFTWARE bifrost` actually works fine, for v0.16.13 and v0.16.14. 
It must just be some weird issue with this particular system.

(BTW, I can see Bifrost has changed quite a bit since I last used it, both visibly and internally. Keep up the good work!)

ethical-haquer, 2024-01-16:
I'm closing this for now, unless I experience this issue on other systems too.

closed 2024-01-16 by ethical-haquer