zacharee/Bifrost #107: 403 Forbidden - All downloading

Martyn911, 2023-09-20:
# Problem with anythink firmwares

#### Example:
Model: **SM-A037G**
Region: **VGR**

[image]

Downloaded file SM-A037G_3_20230907160440_1nnc59kv1j_fac_A037GXXS4CWI1_A037GOXM4CWG7_A037GXXS4CWI1_A037GXXS4CWI1_VGR.zip.enc4 **size 2.2 KB and contains**:

```<!DOCTYPE html>
<html>
	<head>
		<meta charset="utf-8">
		<meta http-equiv="X-UA-Compatible" content="IE=edge">
		<meta name="viewport" content="width=device-width, initial-scale=1">
		<title>403 Forbidden</title>
		<style type="text/css">body{margin:5% auto 0 auto;padding:0 18px}.P{margin:0 22%}.O{margin-top:20px}.N{margin-top:10px}.M{margin:10px 0 30px 0}.L{margin-bottom:60px}.K{font-size:25px;color:#F90}.J{font-size:14px}.I{font-size:20px}.H{font-size:18px}.G{font-size:16px}.F{width:230px;float:left}.E{margin-top:5px}.D{margin:8px 0 0 -20px}.C{color:#3CF;cursor:pointer}.B{color:#909090;margin-top:15px}.A{line-height:30px}.hide_me{display:none}</style>
	</head>
	<body>
		<div id="p" class="P">
			<div class="K">403</div>
			<div class="O I">Forbidden</div>
			<p class="J A L">Error Times: Wed, 20 Sep 2023 06:46:47 GMT
				<br>
				<span class="F">IP: ***</span>Node information: hsh142
				<br>URL: http://cloud-neofussvr.sslcs.cdngc.net/NF_DownloadBinaryForMass.do?file=/neofus/910/SM-A037G_3_20230907160440_1nnc59kv1j_fac.zip.enc4
				<br>Request-Id: 650a9557_hsh142_33960-35094
				<br>
				<br>Check:
				<span class="C G" onclick="s(0)">Details</span></p>
		</div>
		<div id="d" class="hide_me P H">
			<div class="K">ERROR</div>
			<p class="O I">The Requested URL could not be retrieved</p>
			<div class="O">
				<div>While trying to retrieve the URL:</div>
				<pre class="B G">http://cloud-neofussvr.sslcs.cdngc.net/NF_DownloadBinaryForMass.do?file=/neofus/910/SM-A037G_3_20230907160440_1nnc59kv1j_fac.zip.enc4</pre></div>
			<div class="M">
				<span>The following error was encountered:</span>
				<ul class="E">
					<li class="D G">Invalid Request</li></ul>
			</div>
			<p class="M">The access control configuration prevents your request at this time.
				<p></p>Please contact your service provider if you feel this is incorrect.</p>
			<a class="N C" href="#" onclick="s(1)">return</a></div>
		<script type="text/javascript">function e(i) {
				return document.getElementById(i);
			}
			function d(i, t) {
				e(i).style.display = (t ? 'block': 'none');
			}
			function s(e) {
				d('p', e);
				d('d', !e);
			}</script>
	</body>
</html>

BlackMesa123, 2023-09-20:
Same issue when trying to download certain firmwares for specific CSC's, both with [samfirm.js](https://github.com/jesec/samfirm.js) and Bifrost.
- https://github.com/BlackMesa123/proprietary_vendor_samsung_a54x/actions/runs/6235269498
- https://github.com/BlackMesa123/proprietary_vendor_samsung_m52xq/actions/runs/6235308004
Works fine when trying to download the firmware with Frija.

Martyn911, 2023-09-20:
> Same issue when trying to download certain firmwares for specific CSC's, both with [samfirm.js](https://github.com/jesec/samfirm.js) and Bifrost.
>
>     * [...]

Any ideas? Is there no way to intercept traffic from Frija or decompile it? If there is a solution, please let me know.

closed 2023-11-09 by zacharee