zacharee/Bifrost #111: "Bad return status: 400" for  SM-S908B region EUX

divinity76, 2023-10-22:
trying to download SM-S908B with EUX results in
> Bad return status: 400

SM-S908B with EUX definitely exist, here is a screenshot as proof:

[image]
and here is a screenshot of trying to download it with Bifrost:
[image]

interestingly, same happens with EUY but /not/ with BTE (BTE works fine).

divinity76, 2023-10-28:
@Xpl0itR until this is resolved, i wrote a region scanner to find alternative regions at https://github.com/jesec/SamFirm.NET/issues/3

```
$ time php test2.php SM-F916N
total urls to check per model string: 509
total urls to check: 2545
Checking model string 1/5: SM-F916N
http://fota-cloud-dn.ospserver.net/firmware/KTC/SM-F916N/version.xml   : 200
http://fota-cloud-dn.ospserver.net/firmware/LUC/SM-F916N/version.xml   : 200
http://fota-cloud-dn.ospserver.net/firmware/SKC/SM-F916N/version.xml   : 200
http://fota-cloud-dn.ospserver.net/firmware/KOO/SM-F916N/version.xml   : 200
(...)
```
- in my case I couldn't download the correct EUX-region firmware, but my scanner found a compatible region BTE which worked fine
[image]

divinity76, 2023-11-02:
@MichaelDavidGK because my phone's model code is `SM-S908BZWGEUB` , but the update server did not use my full model code, and I wasn't sure exactly what model code the update servers used, so after an exhaustive search,
```
hans@cac3:~$ time php test2.php SM-S908BZWGEUB
total urls to check per model string: 509
total urls to check: 5599
Checking model string 1/11: SM-S908BZWGEUB
Checking model string 2/11: SM-S908BZWGEU
Checking model string 3/11: SM-S908BZWGE
Checking model string 4/11: SM-S908BZWG
Checking model string 5/11: SM-S908BZW
Checking model string 6/11: SM-S908BZ
Checking model string 7/11: SM-S908B
http://fota-cloud-dn.ospserver.net/firmware/AFG/SM-S908B/version.xml   : 200
http://fota-cloud-dn.ospserver.net/firmware/MOB/SM-S908B/version.xml   : 200
http://fota-cloud-dn.ospserver.net/firmware/VIP/SM-S908B/version.xml   : 200
http://fota-cloud-dn.ospserver.net/firmware/O2C/SM-S908B/version.xml   : 200
(...)
```
- it turns out that the update server used the model code `SM-S908B` even though the Samsung warranty page use the model code `SM-S908BZWGEUB`
[image]

> also even if it exists it's a different model, so there is no point to it

I certainly hope not (and the SM-S908B firmware was fully compatible with my SM-S908BZWGEUB model.)

zacharee, 2023-11-10:
There isn't really much I can do if Samsung isn't serving the firmware for a certain CSC directly. As suggested above, you'll have to try a different CSC that contains yours.

divinity76, 2023-11-10:
@zacharee problem is, the Samsung phone itself can download  SM-S908B region EUX, but not Bifrost - The phone is doing /something/ differently, and the server is reacting to it

zacharee, 2023-11-10:
OTA packages use a different server and require a pretty complex registration process that's unique per-device.

divinity76, 2023-11-10:
@zacharee oh ok. close this issue then?

Btw would be nice if Bifrost itself could get a region scanner (I don't do Kotlin myself, but I wrote one in PHP to get around the  SM-S908B+EUX issue - would have saved me the trouble if Bifrost had one)

zacharee, 2023-11-11:
Can you request that in a new issue so I can track it better?

closed 2023-11-11 by zacharee

MK73DS, 2024-09-18:
I experience the same problem with Bifrost 1.19.3 when trying to download S928BXXS3AXI1/S928BOXM3AXI1/S928BXXS3AXI1/S928BXXS3AXI1 for SM-S928B and EUX CSC. The update definitely exists, I got an OTA notification for it and it is present in samfw https://samfw.com/firmware/SM-S928B/EUX/S928BXXS3AXI1
[image]