zacharee/Bifrost #115: I am unable to download the following model: SC51Aa

jianfei96, 2023-11-11:
This is an actual existing model.
Samsung's website.: https://doc.samsungmobile.com/SC51Aa/DCM/doc.html

Could this be due to the tool automatically converting lowercase letters to uppercase?
Note: I want to download SC51Aa, not SC-51A.(Yes, I can download SC-51A)
SC51Aa
[image]
SC-51A
[image]

zacharee, 2023-11-11:
The next version will have an option to allow lowercase letters, but it looks like Samsung isn't serving this firmware anyway: 
[image]

zacharee, 2023-11-12:
1.15.0 has the option (click the settings gear in the bottom right) to allow lowercase characters.

closed 2023-11-12 by zacharee

jianfei96, 2023-11-12:
> The next version will have an option to allow lowercase letters, but it looks like Samsung isn't serving this firmware anyway: [image]

Thanks.
I downloaded it use this tool, with the command "SamFirm.exe -model SC51Aa -region DCM".
https://github.com/ivanmeler/SamFirm_Reborn

So I think it exists.

zacharee, 2023-11-12:
OK, I think this should be fixed in 1.15.1.

For some reason, the firmware needs to be passed all-uppercase to Samsung's servers when requesting the file, even though it has lowercase characters.

jianfei96, 2023-11-12:
> OK, I think this should be fixed in 1.15.1.
> 
> For some reason, the firmware needs to be passed all-uppercase to Samsung's servers when requesting the file, even though it has lowercase characters.

That's great news, thank you for your efforts in development.