zacharee/Bifrost #119: No ARM64 macOS Native Build?

ThePhoenix576, 2023-11-12:
I saw on the changelog that there is supposed to be an ARM64 Native macOS build. but the pre-built macOS App is an Intel only build

[image]

zacharee, 2023-11-12:
It looked like GitHub had added a small runner for macOS ARM64, but it still isn't available. Native ARM64 builds will be available as soon as GitHub adds that runner.

ThePhoenix576, 2023-11-17:
> It looked like GitHub had added a small runner for macOS ARM64, but it still isn't available. Native ARM64 builds will be available as soon as GitHub adds that runner.

Ah I see. That's fine :) I was just wondering since I thought it's a bug or something. Thanks for checking it!

closed 2023-11-17 by ThePhoenix576