zacharee/Bifrost #123: java.lang.IllegalStateException: Could not load font

Vosjedev, 2023-12-10:
I am running archlinux, and I downloaded the latest release.

It complains about not being able to load a font, not showing what font. I installed by downloading the tar.gz, extracting, cd'ing into the directory, and running `bin/bifrost`.

Here is the stacktrace (all output):
<details>
<summary>stacktrace (click to expand)</summary>
```
Dec 10, 2023 6:42:43 PM java.util.prefs.FileSystemPreferences$6 run
WARNING: Prefs file removed in background /home/vosje/.java/.userPrefs/prefs.xml
java.lang.IllegalStateException: Could not load font
        at androidx.compose.ui.text.font.TypefaceRequestCache.runCached(FontFamilyResolver.kt:200)
        at androidx.compose.ui.text.font.FontFamilyResolverImpl.resolve(FontFamilyResolver.kt:92)
        at androidx.compose.ui.text.font.FontFamilyResolverImpl.resolve-DPcqOEQ(FontFamilyResolver.kt:79)
        at androidx.compose.ui.text.platform.ComputedStyle.toSkTextStyle(SkiaParagraph.skiko.kt:227)
        at androidx.compose.ui.text.platform.ParagraphBuilder$makeSkTextStyle$1.invoke(SkiaParagraph.skiko.kt:548)
        at androidx.compose.ui.text.platform.ParagraphBuilder$makeSkTextStyle$1.invoke(SkiaParagraph.skiko.kt:547)
        at androidx.compose.ui.text.WeakKeysCache.get(Cache.jvm.kt:24)
        at androidx.compose.ui.text.platform.ParagraphBuilder.makeSkTextStyle(SkiaParagraph.skiko.kt:547)
        at androidx.compose.ui.text.platform.ParagraphBuilder.textStyleToParagraphStyle(SkiaParagraph.skiko.kt:513)
        at androidx.compose.ui.text.platform.ParagraphBuilder.build(SkiaParagraph.skiko.kt:320)
        at androidx.compose.ui.text.platform.ParagraphLayouter.layoutParagraph(ParagraphLayouter.skiko.kt:162)
        at androidx.compose.ui.text.platform.SkiaParagraphIntrinsics.<init>(SkiaParagraphIntrinsics.skiko.kt:80)
        at androidx.compose.ui.text.platform.SkiaParagraphIntrinsics_skikoKt.ActualParagraphIntrinsics(SkiaParagraphIntrinsics.skiko.kt:37)
        at androidx.compose.ui.text.ParagraphIntrinsicsKt.ParagraphIntrinsics(ParagraphIntrinsics.kt:98)
        at androidx.compose.ui.text.MultiParagraphIntrinsics.<init>(MultiParagraphIntrinsics.kt:102)
        at androidx.compose.foundation.text.modifiers.MultiParagraphLayoutCache.setLayoutDirection(MultiParagraphLayoutCache.kt:250)
        at androidx.compose.foundation.text.modifiers.MultiParagraphLayoutCache.layoutText-K40F9xA(MultiParagraphLayoutCache.kt:275)
        at androidx.compose.foundation.text.modifiers.MultiParagraphLayoutCache.intrinsicHeight(MultiParagraphLayoutCache.kt:203)
        at androidx.compose.foundation.text.modifiers.TextAnnotatedStringNode.maxIntrinsicHeight(TextAnnotatedStringNode.kt:355)
        at androidx.compose.ui.node.LayoutModifierNodeCoordinator.maxIntrinsicHeight(LayoutModifierNodeCoordinator.kt:151)
        at androidx.compose.ui.node.NodeMeasuringIntrinsics$DefaultIntrinsicMeasurable.measure-BRTryo0(LayoutModifierNode.kt:262)
        at androidx.compose.ui.graphics.SimpleGraphicsLayerModifier.measure-3p2s80s(GraphicsLayerModifier.kt:646)
        at androidx.compose.ui.node.LayoutModifierNode$maxIntrinsicHeight$1.measure-3p2s80s(LayoutModifierNode.kt:122)
        at androidx.compose.ui.node.NodeMeasuringIntrinsics.maxHeight$ui(LayoutModifierNode.kt:239)
        at androidx.compose.ui.node.LayoutModifierNode.maxIntrinsicHeight(LayoutModifierNode.kt:120)
        at androidx.compose.ui.node.LayoutModifierNodeCoordinator.maxIntrinsicHeight(LayoutModifierNodeCoordinator.kt:151)
        at androidx.compose.ui.node.LayoutNodeLayoutDelegate$MeasurePassDelegate.maxIntrinsicHeight(LayoutNodeLayoutDelegate.kt:726)
        at androidx.compose.ui.layout.DefaultIntrinsicMeasurable.measure-BRTryo0(Layout.kt:322)
        at androidx.compose.foundation.layout.BoxKt$boxMeasurePolicy$1.measure-3p2s80s(Box.kt:114)
        at androidx.compose.ui.layout.MeasurePolicy.maxIntrinsicHeight(MeasurePolicy.kt:159)
        at androidx.compose.ui.node.IntrinsicsPolicy.maxIntrinsicHeight(IntrinsicsPolicy.kt:49)
        at androidx.compose.ui.node.InnerNodeCoordinator.maxIntrinsicHeight(InnerNodeCoordinator.kt:142)
        at androidx.compose.ui.node.NodeMeasuringIntrinsics$DefaultIntrinsicMeasurable.measure-BRTryo0(LayoutModifierNode.kt:262)
        at androidx.compose.foundation.layout.PaddingNode.measure-3p2s80s(Padding.kt:397)
        at androidx.compose.ui.node.LayoutModifierNode$maxIntrinsicHeight$1.measure-3p2s80s(LayoutModifierNode.kt:122)
        at androidx.compose.ui.node.NodeMeasuringIntrinsics.maxHeight$ui(LayoutModifierNode.kt:239)
        at androidx.compose.ui.node.LayoutModifierNode.maxIntrinsicHeight(LayoutModifierNode.kt:120)
        at androidx.compose.ui.node.LayoutModifierNodeCoordinator.maxIntrinsicHeight(LayoutModifierNodeCoordinator.kt:151)
        at androidx.compose.ui.node.LayoutNodeLayoutDelegate$MeasurePassDelegate.maxIntrinsicHeight(LayoutNodeLayoutDelegate.kt:726)
        at androidx.compose.ui.layout.DefaultIntrinsicMeasurable.measure-BRTryo0(Layout.kt:322)
        at androidx.compose.material3.TabKt$TabBaselineLayout$2$1.measure-3p2s80s(Tab.kt:323)
        at androidx.compose.ui.layout.MeasurePolicy.maxIntrinsicHeight(MeasurePolicy.kt:159)
        at androidx.compose.ui.node.IntrinsicsPolicy.maxIntrinsicHeight(IntrinsicsPolicy.kt:49)
        at androidx.compose.ui.node.InnerNodeCoordinator.maxIntrinsicHeight(InnerNodeCoordinator.kt:142)
        at androidx.compose.ui.node.LayoutNodeLayoutDelegate$MeasurePassDelegate.maxIntrinsicHeight(LayoutNodeLayoutDelegate.kt:726)
        at androidx.compose.foundation.layout.IntrinsicMeasureBlocks$VerticalMaxHeight$1$1.invoke(RowColumnImpl.kt:525)
        at androidx.compose.foundation.layout.IntrinsicMeasureBlocks$VerticalMaxHeight$1$1.invoke(RowColumnImpl.kt:523)
        at androidx.compose.foundation.layout.RowColumnImplKt.intrinsicMainAxisSize(RowColumnImpl.kt:566)
        at androidx.compose.foundation.layout.RowColumnImplKt.intrinsicSize(RowColumnImpl.kt:544)
        at androidx.compose.foundation.layout.RowColumnImplKt.access$intrinsicSize(RowColumnImpl.kt:1)
        at androidx.compose.foundation.layout.IntrinsicMeasureBlocks$VerticalMaxHeight$1.invoke(RowColumnImpl.kt:523)
        at androidx.compose.foundation.layout.IntrinsicMeasureBlocks$VerticalMaxHeight$1.invoke(RowColumnImpl.kt:522)
        at androidx.compose.foundation.layout.RowColumnImplKt$rowColumnMeasurePolicy$1.maxIntrinsicHeight(RowColumnImpl.kt:125)
        at androidx.compose.ui.node.IntrinsicsPolicy.maxIntrinsicHeight(IntrinsicsPolicy.kt:49)
        at androidx.compose.ui.node.InnerNodeCoordinator.maxIntrinsicHeight(InnerNodeCoordinator.kt:142)
        at androidx.compose.ui.node.NodeMeasuringIntrinsics$DefaultIntrinsicMeasurable.measure-BRTryo0(LayoutModifierNode.kt:262)
        at androidx.compose.foundation.layout.FillNode.measure-3p2s80s(Size.kt:698)
        at androidx.compose.ui.node.LayoutModifierNode$maxIntrinsicHeight$1.measure-3p2s80s(LayoutModifierNode.kt:122)
        at androidx.compose.ui.node.NodeMeasuringIntrinsics.maxHeight$ui(LayoutModifierNode.kt:239)
[... 140 lines cut ...]
        at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:773)
        at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:720)
        at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:714)
        at java.base/java.security.AccessController.doPrivileged(AccessController.java:399)
        at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:86)
        at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:742)
        at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203)
        at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124)
        at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113)
        at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109)
        at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
        at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90)
Caused by: java.util.NoSuchElementException: Array is empty.
        at kotlin.collections.ArraysKt___ArraysKt.first(_Arrays.kt:1012)
        at androidx.compose.ui.text.platform.FontCache.loadPlatformTypes-RetOiIg$ui_text(PlatformFont.skiko.kt:173)
[... cut ...]
        at androidx.compose.ui.text.font.FontFamilyResolverImpl$resolve$result$1.invoke(FontFamilyResolver.kt:98)
        at androidx.compose.ui.text.font.FontFamilyResolverImpl$resolve$result$1.invoke(FontFamilyResolver.kt:92)
        at androidx.compose.ui.text.font.TypefaceRequestCache.runCached(FontFamilyResolver.kt:185)
        ... 210 more
Error: The application has experienced an internal error and must quit.
java.lang.IllegalStateException: Could not load font   @ androidx.compose.ui.text.font.TypefaceRequestCache.runCached(FontFamilyResolver.kt:200)
```
</details>

zacharee, 2023-12-18:
Can you tell me:
- Which version of Arch are you using?
- Which window manager are you running?
- Which desktop environment are you using?

Vosjedev, 2023-12-19:
> Which version of Arch are you using?

The latest, I installed a couple of months ago, but kept updateing since then. As far as I know arch does not have os versions the way ubuntu/debian/fedora for example has them, the only version is the ISO.

> Which window manager are you running?

hyprland 0.33.1-2

> Which desktop environment are you using?

 None.

zacharee, 2024-01-15:
Trying on Arch with KDE, Bifrost launches and runs just fine.

I haven't been able to get an instance of Hyprland working to test. Arch base + manual install crashes randomly and silently fails to launch Bifrost. ArcolinuxB boots to a black screen on the live image.

I don't think this is something I can fix, unfortunately.

closed 2024-01-15 by zacharee

Vosjedev, 2024-01-16:
Wel, thanks for attempting to reproduce, than. I guess I will find out what's wrong myself. What fonts are used? I suppose one is missing.

zacharee, 2024-01-16:
It looks like it's looking for either "Noto Sans" or "DejaVu Sans" on Linux.

zacharee, 2024-01-16:
I'm going to try adding my own mapping to fall back to Arial for sans-serif, Times New Roman for serif, and Consolas for monospace, since those aren't part of what Compose looks for on Linux by default, and apparently it doesn't degrade gracefully for missing fonts.

Vosjedev, 2024-01-16:
Is it not posible to specify `sans-serif`, `serif`, and `mono` as fonts to let the system decide what it has like in CSS?

zacharee, 2024-01-16:
Not that I can tell. Generic font families are narrowed down to specific lists depending on the platform inside of Compose. I'm not sure if this is a limitation of Skia, Skiko, Compose, or Java.