zacharee/Bifrost #127: Nothing works today

Fede2782, 2023-12-26:
I tried downloading many firmwares today and I always get this error no matter which device or CSC I'm trying to download. 
Tried on Windows and Android. 

```
java.lang.Exception: java.lang.Exception: Bad return status: 408
        at tk.zwander.commonCompose.view.pages.DownloadViewKt.onDownload(DownloadView.kt:77)
        at tk.zwander.commonCompose.view.pages.DownloadViewKt.access$onDownload(DownloadView.kt:1)
        at tk.zwander.commonCompose.view.pages.DownloadViewKt$onDownload$1.invokeSuspend(DownloadView.kt)
        at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
        at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:108)
        at kotlinx.coroutines.internal.LimitedDispatcher$Worker.run(LimitedDispatcher.kt:115)
        at kotlinx.coroutines.scheduling.TaskImpl.run(Tasks.kt:103)
        at kotlinx.coroutines.scheduling.CoroutineScheduler.runSafely(CoroutineScheduler.kt:584)
        at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.executeTask(CoroutineScheduler.kt:800)
        at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.runWorker(CoroutineScheduler.kt:704)
        at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.run(CoroutineScheduler.kt:691)
Caused by: java.lang.Exception: Bad return status: 408
        at tk.zwander.common.tools.Request.getBinaryFile(Request.kt:198)
        at tk.zwander.common.tools.Request$getBinaryFile$1.invokeSuspend(Request.kt)
        ... 8 more
tk.zwander.common.exceptions.DownloadError: {request=<FUSMsg><FUSHdr><ProtoVer>1.0</ProtoVer><SessionID>0</SessionID><MsgID>1</MsgID></FUSHdr><FUSBody><Put><ACCESS_MODE><Data>2</Data></ACCESS_MODE><BINARY_NATURE><Data>1</Data></BINARY_NATURE><CLIENT_PRODUCT><Data>Smart Switch</Data></CLIENT_PRODUCT><CLIENT_VERSION><Data>4.3.23123_1</Data></CLIENT_VERSION><DEVICE_IMEI_PUSH><Data>12345678901234</Data></DEVICE_IMEI_PUSH><DEVICE_PLATFORM><Data>Android</Data></DEVICE_PLATFORM><DEVICE_FW_VERSION><Data>P615XXS7FWK1/P615OXM7FWJ1/P615XXS7FWK1/P615XXS7FWK1</Data></DEVICE_FW_VERSION><DEVICE_LOCAL_CODE><Data>ITV</Data></DEVICE_LOCAL_CODE><DEVICE_AID_CODE><Data>ITV</Data></DEVICE_AID_CODE><DEVICE_MODEL_NAME><Data>SM-P615</Data></DEVICE_MODEL_NAME><LOGIC_CHECK><Data>X7WWKK676SXW5PX5</Data></LOGIC_CHECK><DEVICE_CONTENTS_DATA_VERSION><Data>P615XXS7FWK1</Data></DEVICE_CONTENTS_DATA_VERSION><DEVICE_CSC_CODE2_VERSION><Data>P615OXM7FWJ1</Data></DEVICE_CSC_CODE2_VERSION><DEVICE_PDA_CODE1_VERSION><Data>P615XXS7FWK1</Data></DEVICE_PDA_CODE1_VERSION><DEVICE_PHONE_FONT_VERSION><Data>P615XXS7FWK1</Data></DEVICE_PHONE_FONT_VERSION><CLIENT_LANGUAGE><Type>String</Type><Type>ISO 3166-1-alpha-3</Type><Data>1033</Data></CLIENT_LANGUAGE></Put><Get><CmdID>2</CmdID><LATEST_FW_VERSION/></Get></FUSBody></FUSMsg>, response=<FUSMsg>
        <FUSHdr>
                <ProtoVer>1.0</ProtoVer>
                <SessionID/>
                <MsgID>1</MsgID>
        </FUSHdr>
        <FUSBody>
                <Results>
                        <CmdRef>2</CmdRef>
                        <Status>408</Status>
                </Results>
        </FUSBody>
</FUSMsg>}
        at tk.zwander.commonCompose.view.pages.DownloadViewKt.onDownload(DownloadView.kt:79)
        at tk.zwander.commonCompose.view.pages.DownloadViewKt.access$onDownload(DownloadView.kt:1)
        at tk.zwander.commonCompose.view.pages.DownloadViewKt$onDownload$1.invokeSuspend(DownloadView.kt)
        at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
        at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:108)
        at kotlinx.coroutines.internal.LimitedDispatcher$Worker.run(LimitedDispatcher.kt:115)
        at kotlinx.coroutines.scheduling.TaskImpl.run(Tasks.kt:103)
        at kotlinx.coroutines.scheduling.CoroutineScheduler.runSafely(CoroutineScheduler.kt:584)
        at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.executeTask(CoroutineScheduler.kt:800)
        at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.runWorker(CoroutineScheduler.kt:704)
        at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.run(CoroutineScheduler.kt:691)
Caused by: java.lang.Exception: Bad return status: 408
        at tk.zwander.common.tools.Request.getBinaryFile(Request.kt:198)
        at tk.zwander.common.tools.Request$getBinaryFile$1.invokeSuspend(Request.kt)
        ... 8 more

```

zacharee, 2023-12-26:
Samsung made it so firmware downloads need to be requested with an IMEI or serial that matches the firmware. 1.16.4 had a workaround to use a value that worked for all models, but Samsung has removed the ability to use that.

I've added a field in 1.16.5 so the IMEI/serial can be specified.

Fede2782, 2023-12-26:
> Samsung made it so firmware downloads need to be requested with an IMEI or serial that matches the firmware. 1.16.4 had a workaround to use a value that worked for all models, but Samsung has removed the ability to use that.
> 
> I've added a field in 1.16.5 so the IMEI/serial can be specified.

Is Samsung serious? If yes I will switch to Pixel forever and goodbye forever Samsung

closed 2023-12-27 by zacharee