zacharee/Bifrost #131: [INFO REQUIRED] Bifrost able to remember the device and CSC details even after its folder is deleted

Dusk-Golem, 2023-12-29:
Yo, as the title suggests, how is Bifrost able to recall the device model and CSC from the previous session even if its folder has been deleted at the session end? I downloaded the firmware for my device SM-A530F and then deleted the Bifrost folder. Even after restaring the PC and extracting Bifrost zip to another folder, it can still recall my device model and the CSC region. WHY? What is the logic for the program to be portable if it is going to leave traces behind once its folder is deleted? Where is the information stored?

zacharee, 2023-12-29:
Bifrost uses the Java Preferences API on desktop.

On Windows, settings are stored in the registry under `HKEY_CURRENT_USER\Software\JavaSoft\Prefs`.

On macOS and Linux, settings are stored under `~/.java/userPrefs/`.

It was a popular request to add this feature.

Dusk-Golem, 2023-12-29:
> Bifrost uses the Java Preferences API on desktop.
> 
> On Windows, settings are stored in the registry under `HKEY_CURRENT_USER\Software\JavaSoft\Prefs`.
> 
> On macOS and Linux, settings are stored under `~/.java/userPrefs/`.
> 
> It was a popular request to add this feature.

I am not against the feature mate, I am just saying that the most logical way to implement it is to create, for example, a config file in the same directory as the program executeable. Or you could add a toggle in Bifrost's settings whether to preserve the aforementioned details for future use or to delete them at program exit.

Anyway, I think it would be easier to add the toggle option instead of removing the functionality altogther so I request you to kindly add it in a future release.

zacharee, 2023-12-29:
The next version will have a button on the "More" page to clear saved data.

Dusk-Golem, 2023-12-29:
Thanks a lot mate. Looking forward to it.

closed 2023-12-29 by zacharee