zacharee/Bifrost #132: Bad return status 408

update-freak, 2023-12-31:
**Describe the bug**
When I try to download the Firmware for SM-A520F with DBT I got the error "Bad return status: 408"
Link: https://www.sammobile.com/samsung/galaxy-s7/firmware/SM-A520F/DBT/

**To Reproduce**
1. Insert  SM-A520F, DBT
2. Click on 'Check for Updates' and then 'Download'

**Expected behavior**
Download should work

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Bifrost version: 1.16.10

Martyn911, 2023-12-31:
Use the service https://samfirm.shop/search/SM-A520F+DBT

Fede2782, 2023-12-31:
You need the IMEI as described in the README. But it is a very old model and Samsung may also stopped serving the fw for that device

update-freak, 2023-12-31:
Ah, sry I forgot the IMEI.
Now it works.
Thanks for the quick response.

closed 2023-12-31 by update-freak