zacharee/Bifrost #137: Program runs with GUI without text.

somerandom907, 2024-01-19:
**Describe the bug**
When I run the program it runs the GUI with no text, no errors in the command prompt. It used to before but after reinstalling windows 10 it doesn't.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'bifrost-1.16.14-windows-amd64\bin\'
2. Click on 'Bifrost.exe'

**Expected behavior**
GUI is supposed to show with text on textboxes, buttons etc.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Windows 10 Build 19045
 - Graphics Card: ATI Mobility Radeon HD 4650
 - Processor: Intel Core2 Duo T6400

zacharee, 2024-01-20:
Does this work? https://github.com/zacharee/SamloaderKotlin/releases/tag/1.16.15-test

somerandom907, 2024-01-20:
Unfortunately it doesn't.
[image]

zacharee, 2024-01-20:
Do you have the ATI drivers installed?

somerandom907, 2024-01-20:
> Do you have the ATI drivers installed?
Yes I do.
[image]

zacharee, 2024-01-20:
Does 1.16.3 work or are all versions not rendering text?

somerandom907, 2024-01-20:
> Does 1.16.3 work or are all versions not rendering text?

If i remember correctly, a few months ago (probably September, October maybe even before) the text was fine it worked. But from like November-December it doesn't. I didn't report it because i thought someone will catch it.
The text is still gone on 1.16.3

[image]

zacharee, 2024-01-20:
Is there a version that does work?

somerandom907, 2024-01-20:
Well comparing the upload dates of the releases, versions 1.13 worked. But when I tried it, it didn't.
I also tried 1.12 for good measure, it also didn't work.

zacharee, 2024-01-20:
The error isn't related. It's just a warning that's fixed in later versions. I think something is broken with your installation, but I don't know what it could be.

zacharee, 2024-01-20:
Actually, are you keeping the folder structure when you extract the ZIP? `Bifrost.exe` should be inside a `bin` folder.

somerandom907, 2024-01-20:
> Actually, are you keeping the folder structure when you extract the ZIP? `Bifrost.exe` should be inside a `bin` folder.

In newer versions yes the exe is in a 'bin' folder, in older ones the exe is in the Bifrost folder.

somerandom907, 2024-01-20:
Should I just reinstall windows?

zacharee, 2024-01-20:
I think you might need to.

somerandom907, 2024-01-20:
Well, I guess there is nothing left to do except that. I'll inform you on if it's working.

somerandom907, 2024-01-21:
So, I reinstalled Windows and it still doesn't work.
[image]

zacharee, 2024-01-22:
I'm curious if there is actually text, but the color is the same as the background for some reason. If you type something in one of the text fields and select it, can you then see the text?

somerandom907, 2024-01-22:
> I'm curious if there is actually text, but the color is the same as the background for some reason. If you type something in one of the text fields and select it, can you then see the text?

No, I cannot.

[image]

zacharee, 2024-01-22:
Can you try this test release? It's forcing Arial to be used instead of looking for Segoe UI.

https://github.com/zacharee/SamloaderKotlin/releases/tag/1.17.2-font-test

somerandom907, 2024-01-22:
> Can you try this test release? It's forcing Arial to be used instead of looking for Segoe UI.
> 
> https://github.com/zacharee/SamloaderKotlin/releases/tag/1.17.2-font-test

Still nothing
[image]
[image]

zacharee, 2024-01-22:
How about this https://github.com/zacharee/SamloaderKotlin/releases/tag/1.17.2-software-render

zacharee, 2024-01-22:
Interesting. That means something is wrong with OpenGL on your machine. What version of the graphics drivers do you have installed?

somerandom907, 2024-01-22:
[image]

somerandom907, 2024-01-22:
This is the latest official driver for my graphics card. There is a beta 13.4 driver but i don't trust beta drivers.

somerandom907, 2024-01-22:
[image]

somerandom907, 2024-01-22:
Yes, the driver is for Win 8 because drivers for Windows 10 doesn't exist. But i used it before and i didn't have any problems like this.

somerandom907, 2024-01-22:
Should I test the beta driver?

zacharee, 2024-01-22:
It looks like the Windows Update Catalog has much newer drivers than what's listed on AMD's website. I'd try letting Windows Update install device drivers. Or you could grab the CAB manually: https://catalog.s.download.windowsupdate.com/c/msdownload/update/driver/drvs/2021/05/display_daf38d1d675861a23a5e5bd3ae052a1d51751ea6.cab.

somerandom907, 2024-01-22:
> It looks like the Windows Update Catalog has much newer drivers than what's listed on AMD's website. I'd try letting Windows Update install device drivers. Or you could grab the CAB manually: [...]

Okay, I will try, but I don't really trust them.

somerandom907, 2024-01-22:
[image]
The driver doesn't work. It's incompatible.

somerandom907, 2024-01-22:
I'll test the beta driver.

zacharee, 2024-01-22:
What version of Windows 10 are you running? That package should work on 1903 and later, which is pretty old.

somerandom907, 2024-01-22:
> What version of Windows 10 are you running? That package should work on 1903 and later, which is pretty old.
> 
> There's also this driver from 2015 https://catalog.s.download.windowsupdate.com/d/msdownload/update/driver/drvs/2016/02/20813855_87923c8bb67e2128747f97506937420fe3835e06.cab.

Windows 10 22H2

somerandom907, 2024-01-22:
How do I install this driver, the .cab file is full of .inf and .cat files?

zacharee, 2024-01-22:
Right click the .inf files and choose "Install".

somerandom907, 2024-01-22:
Okay, thank you.

somerandom907, 2024-01-22:
I did it, nothing happened, just a bunch of messages that inform me that the files are installed.

zacharee, 2024-01-22:
> You are receiving this email because you commented on the thread.

Open Device Manager now and check for driver updates on your GPU there.

Zachary Wander

somerandom907, 2024-01-22:
> Open Device Manager now and check for driver updates on your GPU there. Zachary Wander
> […](#)

I was checking the whole time i was installing, nothing. I reloaded it, after installing it, still nothing, just Microsoft Basic Display Adapter.

zacharee, 2024-01-22:
Alright, I’m not sure what’s up with the drivers, but I’m going to be releasing a new version that should hopefully step through rendering engines until it finds a good one.

Zachary Wander

closed 2024-02-12 by zacharee