zacharee/Bifrost #152: Cannot fetch Galaxy Watch 4 Classic Firmware

techyminati, 2024-02-04:
**Describe the bug**
* The tool cannot fetch Galaxy Watch 4 fw

**To Reproduce**
Steps to reproduce the behavior:
1. Go to device model  Set it to- SM-R890
2. Click on Check Updates
3. Update is found, click on download, it doesnt work and throws error 408, when I give correct serial number it throws error 400
4. See error

**Expected behavior**
Firmware should be downloaded

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows 11 Home
 - Browser: Chrome
 - Version 1.17.5

**Additional context**
Nothing  much

zacharee, 2024-02-05:
Samsung doesn't serve full firmware for watches.

closed 2024-02-05 by zacharee

techyminati, 2024-02-05:
@zacharee it does, I'm getting a 1.7GB OTA from Watch 4 Manager

zacharee, 2024-02-05:
The OTA system is completely separate from the system Bifrost uses, and hasn't been successfully reverse engineered yet.

techyminati, 2024-02-05:
Ah okay, anything we can expect or not?

zacharee, 2024-02-05:
Probably not. The system is very complex and seems to depend on generating unique keys based on the secure element in the device to register it.

techyminati, 2024-02-10:
@zacharee I managed to grab ota link via packet capture, but it says it requires a key value pair

techyminati, 2024-02-10:
@zacharee what could be case where link says "Access Denied"?

techyminati, 2024-02-10:
[URL](https://fota-s3-dn.ospserver.net/firmware/INS/SM-R890/4f191755ab53410a90a1b12b76e6f083.bin?px-hash=&px-wid=2777391-WCC240210073551&px-wctime=2024-02-10%2007:35:51&px-cc=INS&px-mn=SM-R890&Policy=eyJTdGF0ZW1lbnQiOiBbeyJSZXNvdXJjZSI6Imh0dHA6Ly9mb3RhLXMzLWRuLm9zcHNlcnZlci5uZXQvZmlybXdhcmUvSU5TL1NNLVI4OTAvNGYxOTE3NTVhYjUzNDEwYTkwYTFiMTJiNzZlNmYwODMuYmluP3B4LWhhc2g9JnB4LXdpZD0yNzc3MzkxLVdDQzI0MDIxMDA3MzU1MSZweC13Y3RpbWU9MjAyNC0wMi0xMCUyMDA3OjM1OjUxJnB4LWNjPUlOUyZweC1tbj1TTS1SODkwKiIsIkNvbmRpdGlvbiI6eyJEYXRlTGVzc1RoYW4iOnsiQVdTOkVwb2NoVGltZSI6MTcxMTAwNjU1NX19fV19&Signature=YUg88rNXIdvkiNrHWu9kE42xcMHDDIgsu~SKlMTJKOFn3Mre9c~2KAjGvu-OoMXfPple8-fD71e73C~ntJbzcCllBkAApHcXBvyQxnZb0tgTsTdvrdzDtHyCbyJfnp4CwznLu5nvyOZha0ewrA4MNhwFpEYwAiyHz5VCfNTHlG-SwHOEsOl4MOoJdUoHREpr99kUfnq96N4ZO5D4F3FvTw2nTi6iQgxmCpBN1eiA8eei1WwTj3iFSArSeRedLQ20ZCuZW8ycr5glGNMldh4pvxlrIUvgg7bngNSCmiCcRIhM8yGaHHp-mdni-ayYDsRIYR86i9FmcClEZIZKVmEWBA__&Key-Pair-Id=APKAIGLPRCHE2BH6RZLQ&px-nb=UGtezEZ854jbmFcvWGxLEA==)

here