zacharee/Bifrost #164: Access Denied error

jakesullyneytiri, 2024-02-16:
**Describe the bug**
When i try to search for an update for S24 Ultra the app just throws Access Denied no matter what i input as serial or IMEI.

**To Reproduce**
1. Fill the model + Region + IMEI/Serial Number
2. Click on Check for updates
3. App throws Access Denied

**Expected behavior**
I expect it to find updates but for me it just gives Access Denied instead, no matter which samsung phone i try.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows
 - Browser none
 - Version none

**Smartphone (please complete the following information):**
 - Device: s24 Ultra
 - OS: Android 14 (OneUi 6.1)
 - Browser: none
 - Version: none

**Additional context**
I also tried typing in a S23 Ultra IMEI/Serial number and model and program remains with Access Denied result.

s3axel, 2024-02-20:
Not sure whether the "/DS" extension to the model name is the issue, searching with "SM-S928B" gives another error - possibly due to the fact that there is no downloadable firmware up for S24 Ultra yet....

jakesullyneytiri, 2024-02-23:
> Not sure whether the "/DS" extension to the model name is the issue, searching with "SM-S928B" gives another error - possibly due to the fact that there is no downloadable firmware up for S24 Ultra yet....

No, that isn't the cause because it gave the same error even on s23 ultra that has already released versions.

zolfo, 2024-03-02:
Same problem here with a Samsung Zfold3 SM-926B/EUX, AccessDenied.

zolfo, 2024-03-02:
> Same problem here with a Samsung Zfold3 SM-926B/EUX, AccessDenied.

Nevermind, I was missing a "F" in the model "SM-F926B", now it's working for me

zacharee, 2024-03-03:
The `/DS` isn't part of the model. It's just `SM-S928B`.

jakesullyneytiri, 2024-03-03:
> The `/DS` isn't part of the model. It's just `SM-S928B`.

well it doesn't matter which i do, with DS or without both gives same error Access Denied.

closed 2024-03-15 by zacharee