zacharee/Bifrost #166: Bad request status : 400

mystic-mady, 2024-02-21:
I try to download firmware for Samsung Galaxy s20 plus (Bts edition) on Bifrost. But I get Bad request status : 400, I think Odinrom source not working. Can you check it. Thank you 
[image]

mystic-mady, 2024-02-23:
> um if that is your IMEI i wouldn't recommend sharing your IMEI number on internet.

Thanks bruh, what the solution for bad request :400?

zacharee, 2024-03-03:
What model and CSC are you entering?

mystic-mady, 2024-03-03:
Model - SM-G985F
CSC - INU or XSP

zacharee, 2024-03-03:
I'm not able to reproduce:

[image]

mystic-mady, 2024-03-03:
I have purple edition BTS version mobile . I want OXB firmware version not
OXM

zacharee, 2024-03-03:
Use the SIO region.

mystic-mady, 2024-03-03:
Thanks, I will!

closed 2024-03-15 by zacharee