zacharee/Bifrost #170: Flash newly downloaded firmware with Heimdall

JoshAtticus, 2024-03-08:
**Is your feature request related to a problem? Please describe.**
It's awesome having a Samsung Firmware downloader as simple as this one, but it would be even cooler if I could then use Heimdall to flash the newly downloaded firmware.

**Describe the solution you'd like**
Would be cool if there was some sort of GUI to use heimdall to flash the newly downloaded firmware. You can get up-to-date versions of it from https://androidfilehost.com/?w=files&flid=304516

**Describe alternatives you've considered**
Maybe Odin? But that wouldn't work great and only on Windows

**Additional context**
Add any other context or screenshots about the feature request here.

zacharee, 2024-03-24:
I might look into making a GUI for Heimdall, but it's not going to be part of Bifrost.

JoshAtticus, 2024-04-12:
You should close this as not planned instead of completed (since it's not completed)

closed 2024-04-12 by zacharee