zacharee/Bifrost #204: Can't use the decrypter tab to decrypt rom

poqdavid, 2024-04-13:
**Describe the bug**
ROM file won't get decrypted when going to the Decrypter tab but works fine when its decrypted automatically 

**To Reproduce**
Steps to reproduce the behavior:
1. Download the ROM
2. Delete the automatically decrypted zip
3. Going to Decrypter tab
4. Trying to decrypt ROM using the same info that was used to download the ROM

**Expected behavior**
Well was expecting to be able to decrypt the ROM as the app does it automatically after the download

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows 11 Pro
 - Browser Chrome
 - Version 23H2 Build 22631.3447

**Smartphone (please complete the following information):**
 - Device: SM-A155F
 - OS: Android 14
 - Browser Stock
 - Version A155FXXS1AXA4

**Additional context**
Not sure if it helps when trying to decrypt using the decrypter it keeps saying `Status for IMEI <code>: 400` but on the download tab it says `Status for IMEI <code>: 200`

zacharee, 2024-04-13:
I'm not able to reproduce this, unfortunately. Using the same model and region with the generated IMEIs, decrypting a downloaded enc4 works fine.

poqdavid, 2024-04-14:
> I'm not able to reproduce this, unfortunately. Using the same model and region with the generated IMEIs, decrypting a downloaded enc4 works fine.

I just tested the new version and it says `Unable to retrieve v4 key null.` when I am in the decrypter tab
but what makes me wonder why the Downloader tab works fine for me like it checks for CRC and then starts decrypting just fine
is there a way for me to see what key Downloader is using to decrypt the file?

zacharee, 2024-04-14:
Are you trying to decrypt while offline? Bifrost needs to retrieve the decryption key from Samsung. 

If you want to decrypt offline, you can enable the new option on the settings page to save a text file containing the key.

poqdavid, 2024-04-14:
> Are you trying to decrypt while offline? Bifrost needs to retrieve the decryption key from Samsung.
> 
> If you want to decrypt offline, you can enable the new option on the settings page to save a text file containing the key.

Damn that worked tyvm and nope I am not doing it offline both cases i am online but that saving key works perfectly fine :)

zacharee, 2024-04-18:
`.tar.md5` has an MD5 hash somewhere in the file, probably after the actual TAR data.

Does online decryption work for you in the decrypt tab with 1.18.3?

poqdavid, 2024-04-19:
> `.tar.md5` has an MD5 hash somewhere in the file, probably after the actual TAR data.
> 
> Does online decryption work for you in the decrypt tab with 1.18.3?

Sorry for the delay been having trouble rooting my device XD and still do
the 1.18.3 version works if you give it the key and if you don't it gives out this error

[error1.log](https://github.com/zacharee/SamloaderKotlin/files/15045533/error1.log)

zacharee, 2024-04-19:
I can't figure out why the Decrypter page would be failing for you. It performs the exact same request as the Downloader page now.

poqdavid, 2024-04-20:
Yah idk either but since the saved key works now its not like a big problem

closed 2024-04-23 by zacharee