zacharee/Bifrost #217: Download only diff from unpacked FW

MK73DS, 2024-05-12:
**Is your feature request related to a problem? Please describe.**
This feature request is not related to a problem I encounter using Bifrost but more a QoL improvement. Downloading 16GB of firmware each time an update is there is quite tedious. I notice on my phone, when using OTA updates, it only downloads a couple hundreds of MB (see screenshot below). If I already have an unpacked firmware downloaded, would it be possible to download only the "update" itself and not the whole firmware like OTA does?

**Describe the solution you'd like**
Being able to download less data when downloading an updated firmware over an already downloaded firmware. 

**Describe alternatives you've considered**
Downloading the update on my phone using the OTA app in the settings and find a way to extract it from the phone? I don't really know if that's possible.

**Additional context**
Screenshot showing how tiny the update is when using OTA updates.
[image]

zacharee, 2024-05-12:
Samsung's system for downloading OTAs is really complicated. It involves the device registering with Samsung's servers using a key generated based on device identifiers, and I don't think it's ever been fully reverse-engineered. The OTA packages are also in a different format compared to the full firmware; to flash an OTA you need to use ADB sideload in Recovery. OTAs also rely on a specific base firmware version to patch. Samsung sometimes serves multiple OTAs for multiple base versions, but in general, an OTA can only be installed over the previous firmware release.

I'm sure it's technically possible to figure out how the OTA system works and generate a patched full firmware for flashing through ODIN, but it's way outside of what I can do.

MK73DS, 2024-05-13:
Thank you for your detailed answer. I did expect this to be a lot harder than it sounds, otherwise it would already have been done.

closed 2024-05-13 by MK73DS