zacharee/Bifrost #223: Unable to run on Windows

clin1234, 2024-05-24:
Got this error:
```
Exception in thread "main" java.lang.NoClassDefFoundError: Could not initialize class jdk.internal.module.SystemModuleFinders$SystemImage
        at java.base/jdk.internal.module.SystemModuleFinders$SystemModuleReader.findImageLocation(SystemModuleFinders.java:429)
        at java.base/jdk.internal.module.SystemModuleFinders$SystemModuleReader.read(SystemModuleFinders.java:483)
        at java.base/jdk.internal.loader.BuiltinClassLoader.defineClass(BuiltinClassLoader.java:809)
        at java.base/jdk.internal.loader.BuiltinClassLoader.findClassInModuleOrNull(BuiltinClassLoader.java:741)
        at java.base/jdk.internal.loader.BuiltinClassLoader.loadClassOrNull(BuiltinClassLoader.java:665)
        at java.base/jdk.internal.loader.BuiltinClassLoader.loadClass(BuiltinClassLoader.java:639)
        at java.base/jdk.internal.loader.ClassLoaders$AppClassLoader.loadClass(ClassLoaders.java:188)
        at java.base/java.lang.ClassLoader.loadClass(ClassLoader.java:521)
Caused by: java.lang.ExceptionInInitializerError: Exception java.lang.UnsatisfiedLinkError: no jimage in system library path: C:\Users\×š×™× ×©×’×›×”×“\scoop\apps\bifrost\current\bin [in thread "main"]
        at java.base/java.lang.ClassLoader.loadLibrary(ClassLoader.java:2408)
        at java.base/java.lang.Runtime.loadLibrary0(Runtime.java:848)
        at java.base/java.lang.System.loadLibrary(System.java:2015)
        at java.base/jdk.internal.jimage.NativeImageBuffer$1.run(NativeImageBuffer.java:42)
        at java.base/jdk.internal.jimage.NativeImageBuffer$1.run(NativeImageBuffer.java:40)
        at java.base/java.security.AccessController.doPrivileged(AccessController.java:318)
        at java.base/jdk.internal.jimage.NativeImageBuffer.<clinit>(NativeImageBuffer.java:39)
        at java.base/jdk.internal.jimage.BasicImageReader.<init>(BasicImageReader.java:97)
        at java.base/jdk.internal.jimage.ImageReader$SharedImageReader.<init>(ImageReader.java:229)
        at java.base/jdk.internal.jimage.ImageReader$SharedImageReader.open(ImageReader.java:243)
        at java.base/jdk.internal.jimage.ImageReader.open(ImageReader.java:67)
        at java.base/jdk.internal.jimage.ImageReader.open(ImageReader.java:71)
        at java.base/jdk.internal.jimage.ImageReaderFactory$1.apply(ImageReaderFactory.java:70)
        at java.base/jdk.internal.jimage.ImageReaderFactory$1.apply(ImageReaderFactory.java:67)
        at java.base/java.util.concurrent.ConcurrentHashMap.computeIfAbsent(ConcurrentHashMap.java:1708)
        at java.base/jdk.internal.jimage.ImageReaderFactory.get(ImageReaderFactory.java:61)
        at java.base/jdk.internal.jimage.ImageReaderFactory.getImageReader(ImageReaderFactory.java:85)
        at java.base/jdk.internal.module.SystemModuleFinders$SystemImage.<clinit>(SystemModuleFinders.java:385)
        ... 8 more
```

zacharee, 2024-05-26:
Is your user folder really named `×š×™× ×©×’×›×”×“`?

clin1234, 2024-05-26:
No, somewhere in the code character encoding really messed up. My actual username on Windows is ךינשגכהד

zacharee, 2024-08-05:
This should be fixed in 1.18.15.

closed 2024-08-05 by zacharee