zacharee/Bifrost #229: Access Denied when click Check Update

hazukame, 2024-06-24:
I'm trying to find the firmware for my device but when I fill everything out it shows Access Denied. I think I've done everything I can from running as administrator to installing everything possible on github, but still nothing. I don't know what this is caused by

[image]

zacharee, 2024-06-24:
Don't include the `/DS`.

hazukame, 2024-06-24:
> Don't include the `/DS`.

As I said, I tried everything - So also adding /DS, even when it does not add it shows the same Access Denied error

[image]

EFIShell0, 2024-06-24:
Just try it as DS. You can also find the code of your device from its box or via Android (such as SM-A105F, SM-X810).

closed 2024-06-24 by hazukame