zacharee/Bifrost #261: Cannot open file selector in Linux (KDE Plasma Wayland)

MK73DS, 2024-08-15:
**Describe the bug**
When downloading a firmware, a window opens to select where to download it. On the latest version (1.19.0) that window is completely black. 

**To Reproduce**
1. Launch Bifrost
2. In the download tab, click download

**Expected behavior**
The app should open a file selector. This used to work in the previous version I used (1.18.9). 

**Screenshots**
Bifrost 1.19.0 
[image]
Bifrost 1.18.9 
[image]

**Desktop (please complete the following information):**
 - OS: Arch Linux (KDE Plasma on Wayland)
 - Version: 1.19.0

kostadinsh, 2024-08-16:
Possibly related to https://github.com/zacharee/SamloaderKotlin/issues/34

The output below gets spammed if the program is ran from the command line:

```

(bifrost:135962): Gtk-WARNING **: 23:37:05.296: drawing failure for widget 'GtkFileChooserDialog': error occurred in libfreetype

(bifrost:135962): Gtk-WARNING **: 23:37:05.313: drawing failure for widget 'GtkTreeView': error occurred in libfreetype

(bifrost:135962): Gtk-WARNING **: 23:37:05.313: drawing failure for widget 'GtkScrolledWindow': error occurred in libfreetype

(bifrost:135962): Gtk-WARNING **: 23:37:05.313: drawing failure for widget 'GtkBox': error occurred in libfreetype

(bifrost:135962): Gtk-WARNING **: 23:37:05.313: drawing failure for widget 'GtkStack': error occurred in libfreetype

(bifrost:135962): Gtk-WARNING **: 23:37:05.313: drawing failure for widget 'GtkBox': error occurred in libfreetype

(bifrost:135962): Gtk-WARNING **: 23:37:05.313: drawing failure for widget 'GtkBox': error occurred in libfreetype

(bifrost:135962): Gtk-WARNING **: 23:37:05.313: drawing failure for widget 'GtkPaned': error occurred in libfreetype
```

zacharee, 2024-08-17:
Can you let me know if 1.19.1 fixes the issue?

kostadinsh, 2024-08-17:
Still broken for me, maybe you also need to add detection if a user is running Plasma 5 or 6? I don't have the executable it's trying to run, and instead I have kreadconfig6

```
Aug 17, 2024 9:27:48 PM java.util.prefs.FileSystemPreferences$1 run
INFO: Created user preferences directory.
Aug 17, 2024 9:27:48 PM java.util.prefs.FileSystemPreferences$6 run
WARNING: Prefs file removed in background /home/kostadin/.java/.userPrefs/prefs.xml
java.io.IOException: Cannot run program "kreadconfig5": error=2, No such file or directory
        at java.base/java.lang.ProcessBuilder.start(ProcessBuilder.java:1170)
        at java.base/java.lang.ProcessBuilder.start(ProcessBuilder.java:1089)
        at java.base/java.lang.Runtime.exec(Runtime.java:681)
        at java.base/java.lang.Runtime.exec(Runtime.java:530)
        at tk.zwander.common.ui.LinuxAccentColorGetterKt.getLinesFromCommand(LinuxAccentColorGetter.kt:86)
        at tk.zwander.common.ui.DESpecificGetter$KDE.getAccentColor-QN2ZGVo(LinuxAccentColorGetter.kt:40)
        at tk.zwander.common.ui.LinuxAccentColorGetter.getAccentColor-QN2ZGVo(LinuxAccentColorGetter.kt:11)
        at tk.zwander.commonCompose.util.ThemeUtilsJVM.rememberThemeInfo(ThemeUtils.kt:65)
        at MainKt$main$1$1.invoke(main.kt:107)
        at MainKt$main$1$1.invoke(main.kt:105)
        at androidx.compose.runtime.internal.ComposableLambdaImpl.invoke(ComposableLambda.jb.kt:107)
        at androidx.compose.runtime.internal.ComposableLambdaImpl.invoke(ComposableLambda.jb.kt:33)
        at androidx.compose.runtime.CompositionLocalKt.CompositionLocalProvider(CompositionLocal.kt:401)
        at MainKt$main$1.invoke(main.kt:103)
        at MainKt$main$1.invoke(main.kt:102)
        at androidx.compose.runtime.internal.ComposableLambdaImpl.invoke(ComposableLambda.jb.kt:116)
        at androidx.compose.runtime.internal.ComposableLambdaImpl.invoke(ComposableLambda.jb.kt:33)
        at androidx.compose.ui.window.Application_desktopKt$application$1$1.invoke(Application.desktop.kt:117)
        at androidx.compose.ui.window.Application_desktopKt$application$1$1.invoke(Application.desktop.kt:116)
        at androidx.compose.runtime.internal.ComposableLambdaImpl.invoke(ComposableLambda.jb.kt:116)
        at androidx.compose.runtime.internal.ComposableLambdaImpl.invoke(ComposableLambda.jb.kt:33)
        at androidx.compose.ui.window.Application_desktopKt$awaitApplication$2$1$2$1$1.invoke(Application.desktop.kt:233)
        at androidx.compose.ui.window.Application_desktopKt$awaitApplication$2$1$2$1$1.invoke(Application.desktop.kt:232)
        at androidx.compose.runtime.internal.ComposableLambdaImpl.invoke(ComposableLambda.jb.kt:107)
        at androidx.compose.runtime.internal.ComposableLambdaImpl.invoke(ComposableLambda.jb.kt:33)
        at androidx.compose.runtime.CompositionLocalKt.CompositionLocalProvider(CompositionLocal.kt:380)
        at androidx.compose.ui.window.Application_desktopKt$awaitApplication$2$1$2$1.invoke(Application.desktop.kt:223)
        at androidx.compose.ui.window.Application_desktopKt$awaitApplication$2$1$2$1.invoke(Application.desktop.kt:221)
        at androidx.compose.runtime.internal.ComposableLambdaImpl.invoke(ComposableLambda.jb.kt:107)
        at androidx.compose.runtime.internal.ComposableLambdaImpl.invoke(ComposableLambda.jb.kt:33)
        at androidx.compose.runtime.ActualJvm_jvmKt.invokeComposable(ActualJvm.jvm.kt:33)
        at androidx.compose.runtime.ComposerImpl.doCompose(Composer.kt:3595)
        at androidx.compose.runtime.ComposerImpl.composeContent$runtime(Composer.kt:3522)
        at androidx.compose.runtime.CompositionImpl.composeContent(Composition.kt:743)
        at androidx.compose.runtime.Recomposer.composeInitial$runtime(Recomposer.kt:1114)
        at androidx.compose.runtime.CompositionImpl.composeInitial(Composition.kt:649)
        at androidx.compose.runtime.CompositionImpl.setContent(Composition.kt:635)
        at androidx.compose.ui.window.Application_desktopKt$awaitApplication$2$1$2.invokeSuspend(Application.desktop.kt:221)
        at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
        at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:101)
        at java.desktop/java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:318)
        at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:773)
        at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:720)
        at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:714)
        at java.base/java.security.AccessController.doPrivileged(AccessController.java:400)
        at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:87)
        at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:742)
        at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203)
        at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124)
        at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113)
        at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109)
        at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
        at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90)
Caused by: java.io.IOException: error=2, No such file or directory
[... cut ...]
(bifrost:233463): Gtk-WARNING **: 21:28:02.853: drawing failure for widget 'GtkGrid': error occurred in libfreetype

(bifrost:233463): Gtk-WARNING **: 21:28:02.853: drawing failure for widget 'GtkBox': error occurred in libfreetype

(bifrost:233463): Gtk-WARNING **: 21:28:02.853: drawing failure for widget 'GtkFileChooserWidget': error occurred in libfreetype

(bifrost:233463): Gtk-WARNING **: 21:28:02.853: drawing failure for widget 'GtkBox': error occurred in libfreetype

(bifrost:233463): Gtk-WARNING **: 21:28:02.853: drawing failure for widget 'GtkFileChooserDialog': error occurred in libfreetype
```

MK73DS, 2024-08-17:
> Can you let me know if 1.19.1 fixes the issue?

No it doesn't, it still displays a black window like in 1.19.0.

zacharee, 2024-08-20:
This should be properly fixed in 1.19.2.

closed 2024-08-21 by MK73DS