zacharee/Bifrost #269: 1.19.2 bifrost

Serg19644, 2024-08-26:
На версии для windows скачивает файл CRC и удаляет. Прошивку не удаётся скачать. На версии 1.19.1 всё нормально работает.

On the windows version, downloads the CRC file and deletes it. The firmware cannot be downloaded. Everything works fine on version 1.19.1.

Serg19644, 2024-08-28:
The program downloads the firmware, but after downloading it checks and deletes it. At the end of the download, the zip file weighs 0 bytes

WesSec, 2024-08-28:
I think it has to do with the save window trying to save it as .zip instantly instead of enc4

zacharee, 2024-08-28:
This should be fixed in 1.19.3, which is building now.

Serg19644, 2024-08-29:
Fixed in version 1.19.3. Everything is working now, thanks.

closed 2024-08-31 by zacharee