zacharee/Bifrost #283: Will there ever be a way to generate SERIALS?

qtqgyt, 2024-09-29:
> Will there ever be a way to generate SERIALS, eg. for getting firmware for tablets?

Actually, I will move this to a new issue.

_Originally posted by @qtqgyt in https://github.com/zacharee/SamloaderKotlin/issues/116#issuecomment-2381629740_

zacharee, 2024-09-29:
IMEIs follow a publicly-defined standard for their format, with only a few digits that are up to the manufacturer to fill in. I don't know how Samsung creates their serial numbers so I can't create a generator.

qtqgyt, 2024-09-29:
That's unfortunate. I was just hoping it was at least a semi-known algorithm. Thanks though!

qtqgyt, 2024-09-29:
I was hoping to make maybe my own firmware archive, so unless I want to go figuring out how they make them, which I would need a large dataset of serials for, I guess I gotta go ebay shopping, or just give up on that idea. Sometimes I have the most random project ideas, so I suddenly get motivated to try and make it, and then it either gets too hard, is impossible, or I just give up on. I don't know why I am like that, I was just hoping that there was a way to do it.

qtqgyt, 2024-09-29:
Maybe you can make a database of serials too!
Anyways, I just was thinking, and that's my random thought. Have a good day!

closed 2024-09-29 by qtqgyt