zacharee/Bifrost #292: Throwing exceptions while running on Windows 7 systems

thisbing, 2024-10-18:
**Describe the bug**
I executed Bifrost\bin\Bifrost.exe and the operating system popped up an error message: java.lang.UnstatisLinkError:error looking up function 'IsWow64Process2'.
The full path to Bifrost is D:\Bifrost\bin\Bifrost.exe with no spaces.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to D:\Bifrost\bin
2. Click on Bifrost.exe
3. throw an exception: java.lang.UnstatisLinkError:error looking up function 'IsWow64Process2'.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
Opens normally and downloads firmware quickly.

**Desktop (please complete the following information):**
 - OS: Microsoft Windows 7 Service Pack 1
 Bifrost Version: 1.19.4

Thank you very much for providing this tool. I hope this problem can be solved as soon as possible. I wish you a happy life and all the best.

[image]
[image]

zacharee, 2024-10-20:
This should be fixed in 1.19.5.

thisbing, 2024-10-21:
Fixed in version 1.19.3. Everything is working now, thanks.

closed 2024-10-21 by thisbing