zacharee/Bifrost #296: Error: Index 2 out of bounds for length 1

Nnamllit13, 2024-10-23:
**Bug Description**
I am currently experiencing the following error: when I want to check a firmware for updates in order to download it, the following error occurs: "Index 2 out of bounds for length 1".

**Desktop:**
 - OS: Windows 10 Pro

**Smartphone Info:**
 - Device: Samsung galaxy a51
 - Model: SM-A515F
 - Region: EUX
 - IMEI: Is the default value <details><summary>Values </summary><p>350220431234566;350475671234565;350608751234564;350678891234568;351017181234567;351547441234569;351824381234564;351919551234566;352237111234565;352248111234562;352248231234567;352249111234560;352333111234562;352334111234560;352349111234568;352350111234566;352351111234564;352352111234562;352353111234560;352354111234568;352355111234565;352617131234564;352768501234567;352885871234561;353048181234560;353194561234562;353303641234563;353327111234563;353646771234567;353678111234569;353679111234567;353680111234565;353681111234563;353682111234561;353684111234567;353685111234564;353686111234562;353687111234560;354001511234569;354030111234562;354106681234560;354455941234567;354464111234563;354465111234560;354468111234564;354470111234560;354471111234568;354558591234561;354561101234568;354561111234566;354562111234564;354563111234562;354564111234560;354595381234567;355035111234560;355036111234568;355037111234566;355041111234568;355042111234566;355078741234568;355199111234564;355203211234563;355255111234564;355256111234562;355462111234566;355463111234564;355707111234564;355708111234562;355773111234566;355815881234563;355851111234560;355998111234569;356057111234563;356161111234563;356162111234561;356333351234567;356617141234568;357112781234564;357510191234563;357560401234564;357641751234568;357659421234560;358338561234560;358719831234565;358760471234564;358960431234569;359054841234564;359062341234562;359178861234566;350220430123455;350475670123454;350608750123453;350678890123457;351017180123456;351547440123458;351824380123453;351919550123455;352237110123454;352248110123451;352248230123456</p></details> 
[image]

```
     </body>
   </html>
```
(Unfortunately I could not capture everything in one screenshot)

Nnamllit13, 2024-11-01:
I am just stupid :)

closed 2024-11-01 by Nnamllit13

grasmanek94, 2025-09-15:
> [@grasmanek94](https://github.com/grasmanek94) "EUX" is a generic term for the European market/firmware versions. If the device is from the Netherlands, you must use "DBE" (if you want the unbranded firmware version). You can find all other CSC codes for the corresponding countries/firmware versions here: https://samfw.com/firmware/SM-G996B

Thanks, I tried a different CSC and that worked indeed.